zephyrproject-rtos/zephyr #18019: BT scan via shell fatal error

chtavares592, 2019-08-05:
**Describe the bug**
I'm testing the sample `samples/bluetooth/central/` and add the `CONFIG_BT_SHELL=y` option to use shell options on the test, but when I active the scan, happens a FATAL error.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b  nrf52840_pca10056  -d build-display
3. west flash
4. See error

**Console output**
````
uart:~$ bt scan off
Scan successfully stopped
uart:~$ bt scan on
Bluetooth active scan enabled
uart:~$ FATAL: ***** BUS FAULT *****
FATAL:   Precise data bus error
FATAL:   BFAR Address: 0x868fb8f
FATAL: r0/a1:  0x200018ec  r1/a2:  0xfffffff8  r2/a3:  0x13f204fa
FATAL: r3/a4:  0x0868fb81 r12/ip:  0x00000000 r14/lr:  0x0001eee1
FATAL:  xpsr:  0x81000000
FATAL: Faulting instruction address (r15/pc): 0x0001d500
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x20000978 (unknown)
FATAL: Halting system

````

**Environment (please complete the following information):**
 - OS: Debian 
 - Toolchain Zephyr SDK  0.10.1
 - Zephyr Master, commit SHA fc79a789c7e518e12a3e0b9e8c62c823adb12d61

aescolar, 2019-08-05:
CC @carlescufi

joerchan, 2019-08-14:
This happens because `bt init` has not been called.
So `ctx_shell` in `found_device` callback has not been initialized, this causes a memory bug in `shell_print(ctx_shell, ...)`

carlescufi, 2019-08-15:
Closing since it's the user's responsibility to call `bt_init()`

closed 2019-08-15 by carlescufi