zephyrproject-rtos/zephyr #18021: Socket vtable can access null pointer callback function

jukkar, 2019-08-05:
**Describe the bug**
If some socket vtable function is not implemented, then the callback pointer would be NULL. This is not checked when calling the callback which can lead to null pointer access.

pfalcon, 2019-08-06:
> If some socket vtable function is not implemented, then the callback pointer would be NULL.

If some socket vtable function is not implemented, it should be stubbed with a dummy function, return an appropriate error. This was a design decision, to avoid avoidable NULL pointer check, and leave a bit more freedom re: how to handle such "not implemented" cases.

jukkar, 2019-08-06:
> If some socket vtable function is not implemented, it should be stubbed with a dummy function, return an appropriate error. This was a design decision, to avoid avoidable NULL pointer check, and leave a bit more freedom re: how to handle such "not implemented" cases.

IMHO this is not good. We end up creating unnecessary dummy functions without any benefit. If you really want to handle "not implemented" case in a special way, then you need to create a dummy function that does something special. If the default handling is fine, then the dummy function can be left unimplemented.

jukkar, 2019-08-06:
The reason for creating this issue, was that I got hit by it as I only wanted to implement couple of callbacks and was not interested in others. Creating dummys in this case would be pointless as the default handling would be just fine.

pfalcon, 2019-08-06:
> We end up creating unnecessary dummy functions without any benefit.

A single one is enough.

pfalcon, 2019-08-06:
> as the default handling would be just fine.

"Default handling" should be "not take deliberate steps to make code slower, if that's easily avoidable". (Yeah, it's all about couple of CPU cycles. And there're much more inefficiency in the code around. That's more about attitude - if we don't care about obvious cases now, then who and when will go and deal with more complex ones? And well, calling a vmethod is pretty much on a critical path in any design involving vmethods, so thinking about it is not useless.)

All in all, my point is that it wasn't an omission, but a deliberate choice in the original impl. I'm not going to either -1 or +1 the corresponding patch to change it, up to other folks to consider pros and cons.

PR #18022 (2019-08-06): net: sockets: Do not call NULL socket callback function
Before calling socket callback function, make sure the callback
function exists so that we do not get NULL pointer reference.

Fixes #18021
commit 0a0fb72d (2019-08-06): net: sockets: Do not call NULL socket callback function

Before calling socket callback function, make sure the callback
function exists so that we do not get NULL pointer reference.

Fixes #18021
commit 764b6685 (2019-08-06): net: sockets: Do not call NULL socket callback function

Before calling socket callback function, make sure the callback
function exists so that we do not get NULL pointer reference.

Fixes #18021
commit abdd58ce (2019-08-06): net: sockets: Do not call NULL socket callback function

Before calling socket callback function, make sure the callback
function exists so that we do not get NULL pointer reference.

Fixes #18021
closed 2019-08-06 by jukkar