zephyrproject-rtos/zephyr #18031: samples/shields/x_nucleo_iks01a3 test is stucking due to dca45cb commit

avisconti, 2019-08-06:
The samples/shields/x_nucleo_iks01a3 is a sample to test the x_nucleo_iks01a3 shield. It outputs the sensor data forever.
After dca45cb commit has been merged it stucks after few iterations
(dca45cb drivers: i2c: Add STM32F10X slave support)
Help on this commit is required.
(see #17568)

If I remove it everything works:
git revert dca45cb

Steps to reproduce the behavior:
(it requires a nucleo F401RE + x_nucleo_iks01a3 shield)
1. samples/shields/x_nucleo_iks01a3/standard
2. mkdir build && cd build
3. cmake -DBOARD=nucleo_f401re ..
4. make

**Screenshots or console output**
X-NUCLEO-IKS01A3 sensor dashboard

HTS221: Temperature: 26.4 C
HTS221: Relative Humidity: 30.0%
LPS22HH: Temperature: 0.0 C
LPS22HH: Pressure:0.000 kpa
STTS751: Temperature: 0.0 C
LIS2MDL: Magn (gauss): x: -0.660, y: -0.351, z: -0.804
LIS2DW12: Accel (m.s-2): x: 43.109, y: -1.838, z: 69.296
LSM6DSO: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000
LSM6DSO: GYro (dps): x: 0.000, y: 0.000, z: 0.000
1:: lps22hh trig 4
1:: lsm6dso acc trig 8
1:: lsm6dso gyr trig 8
1:: lsm6dso temp trig 0
[00:00:00.004,000] <dbg> LIS2DW12.lis2dw12_set_power_mode: Apply default Power Mode
[00:00:00.008,000] <inf> LSM6DSO: chip id 0x6c

avisconti, 2019-08-06:
@erwango @ydamigos 
I put you as owner of this bug as you reviewed #17568.
I have actually been requested by @ydamigos to review it as well, but I was in vacation. Sorry for
that. I think that we should put @pavlohamov in a condition to debug it.

ydamigos, 2019-08-06:
I will take a look at it the following days. Unfortunately, I don't have the hardware to test it.
@avisconti Did you test the sample in polling mode?

pavlohamov, 2019-08-06:
@avisconti Thank you for pointing this issue. Seems like i've managed to reproduce it on stm32f429 disco with 100% probability. 
The issue is NOT reproduced if:
`CONFIG_I2C_STM32_INTERRUPT=n`
or
`CONFIG_I2C_SLAVE=y`
Also with https://github.com/zephyrproject-rtos/zephyr/pull/18061 - works fine.

Unfortunate i have no nucleo shield, nor F401. @avisconti Could you be so kind to verify these 3 individual options?

avisconti, 2019-08-07:
> I will take a look at it the following days. Unfortunately, I don't have the hardware to test it.
> @avisconti Did you test the sample in polling mode?

No, interrupt mode.

avisconti, 2019-08-07:
> @avisconti Thank you for pointing this issue. Seems like i've managed to reproduce it on stm32f429 disco with 100% probability.
> The issue is NOT reproduced if:
> `CONFIG_I2C_STM32_INTERRUPT=n`
> or [...]

@pavlohamov 
I tried #18061 and it seems to fix my issue!!
I will take a deeper look to what it does, and maybe @ydamigos can also review it.
Thanks for the quick solution.

dleach02, 2019-10-15:
@avisconti , can you confirm that the #18061 fixes this issue and you want to have it backported to 2.0.1?

avisconti, 2019-10-16:
> @avisconti , can you confirm that the #18061 fixes this issue and you want to have it backported to 2.0.1?

@dleach02 
David, yes it fixes the issue and I believe it must be backported to v2.0.1.
Thanks!

erwango, 2019-11-21:
@avisconti, can we close this bug ? It is counting in V2.1.0 maturity while it is actually closed on this branch

dleach02, 2019-11-26:
@avisconti, please open an issue item for your backport request. We are going to close this particular issue.

closed 2019-11-26 by dleach02

avisconti, 2019-11-26:
> @avisconti, please open an issue item for your backport request. We are going to close this particular issue.

@erwango @dleach02 
Sorry, I missed your comments.
Sure I will open a new request for v2.0.1