zephyrproject-rtos/zephyr #18034: It's impossible to build Zephyr via cmake/make with west 0.6.0 installed

PiotrZierhoffer, 2019-08-06:
**Describe the bug**
If not cloning using west (but just regular git clone), it's impossible to build with the usual cmake/make commands.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/zephyrproject-rtos/zephyr
2. cd samples/hello_world
3. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained #or any other
3. make
4. See error: `FATAL ERROR: no west installation found from ".../zephyr/samples/hello_world/build"; "west list" requires one.`

aescolar, 2019-08-06:
CC: @mbolivar

carlescufi, 2019-08-06:
`west` is required upstream because of modules. Unless you provide a list of modules separately, you need to have west installed. This is not a bug.

carlescufi, 2019-08-06:
If you really want to build without west then you need to provide CMake with a `ZEPHYR_MODULES` variable: https://docs.zephyrproject.org/latest/guides/modules.html

PiotrZierhoffer, 2019-08-06:
The documentation you pointed to explicitly says it's fine not to use modules:

> If you don’t have west installed and don’t specify ZEPHYR_MODULES yourself, then no additional modules are added to the build. You will still be able to build any applications that don’t require code or Kconfig options defined in an external repository.

PiotrZierhoffer, 2019-08-06:
The main problem comes from the change in the error message that is filtered out in Zephyr build system. Please take a look at the PR I suggested.

carlescufi, 2019-08-06:
> The main problem comes from the change in the error message that is filtered out in Zephyr build system. Please take a look at the PR I suggested.

Yep, sorry for the noise, I confused the root cause of the issue.

PR #18035 (2019-08-06): [#18034] Update the error message when not in west dir
Zephyr verifies if it is built from a west-initialized directory.
The message changed in west@c08061cef1c7b0e19a58a82db731098e2f4bba4a.

Closes #18034.
commit a3ddc09a (2019-08-06): build: west: Update the error message when not in west dir

Zephyr verifies if it is built from a west-initialized directory.
The message changed in west@c08061cef1c7b0e19a58a82db731098e2f4bba4a.

Closes #18034.
closed 2019-08-06 by carlescufi