zephyrproject-rtos/zephyr #18042: Only corporate members can join the slack channel

olofj, 2019-08-06:
Right now it seems that only people with certain corporate email addresses can join the slack channel. Given that slack is communicated as an official support channel, this seems like a bug.

At the very least, it should have information to how you get invited to the workspace if you have a Slack account under another email address.

Current login page says:

If you have an @intel.com, @linaro.com, @linux.intel.com, @nordicsemi.no, @nxp.com, @oticon.com, @foundries.io, @ti.com, @synopsys.com, @antmicor.com, or @sifive.com email address, you can create an account.

carlescufi, 2019-08-06:
I fixed this yesterday I believe, see the invite link in the GitHub doc: https://tinyurl.com/y5glwylp I also updated the link in the doc, it might not be updated yet on the website

olofj, 2019-08-06:
Fantastic, thanks!

closed 2019-08-06 by olofj

scttnlsn, 2020-05-01:
It seems as though this invite link is no longer working:

```
Secure Connection Failed

An error occurred during a connection to tinyurl.com. PR_END_OF_FILE_ERROR

    The page you are trying to view cannot be shown because the authenticity of the received data could not be verified.
    Please contact the website owners to inform them of this problem.
```

schneider42, 2021-09-15:
The link posted above is also on the website: https://www.zephyrproject.org/contact-us/

It does lead to slack, but the invitation is not working:

[image]

schneider42, 2021-09-15:
Never mind: Discord is the new Slack: https://lists.zephyrproject.org/g/users/message/2702