zephyrproject-rtos/zephyr #18047: BT: Advertising Extensions - Advertiser #94

aescolar, 2019-08-06:
The host needs to support the HCI commands related to Advertising Extensions as an advertiser. If talking to a controller with support these will have to be used instead of the Bluetooth 4.0 commands for advertising

This can be upstreamed as it is in already adopted spec. (Bluetooth Core Specification 5.0)

dleach02, 2020-01-11:
We seem to keep moving this down the release chain. Is this something that is actively being worked?

jhedberg, 2020-01-14:
> We seem to keep moving this down the release chain. Is this something that is actively being worked?

@dleach02 AFAIK @joerchan has been working on it and should be submitting a draft PR soon. Actually, should this be reassigned from @carlescufi to @joerchan ?

joerchan, 2020-01-14:
Assigned myself and put them in progress.

carlescufi, 2020-03-10:
Covered by #18050

closed 2020-03-10 by carlescufi