zephyrproject-rtos/zephyr #18048: [zephyr 1.14] Zephyr with privacy does not disconnect device with unresolvable RPA

MariuszSkamra, 2019-08-06:
**Describe the bug**
Remote performs pairing with Zephyr and distributes IRK. After disconnection it sets it's address to unresolvable RPA using invalid IRK (different than previously distributed). 

According to the Core Specification 5.1 | Vol 3, Part C, 10.7.1 Privacy feature in a Peripheral
```
If the resolution procedure fails, then the Host shall disconnect with
the error code "Authentication failure", or perform the pairing procedure, or
perform the authentication procedure as defined in Section 10.3.
```
The test case GAP/PRIV/CONN/BI-01-C requires Zephyr to disconnect the link with “Authentication Failure” error code.

**To Reproduce**
Steps to reproduce the behavior:
1. Run GAP/PRIV/CONN/BI-01-C with btshell application

**Expected behavior**
Zephyr disconnects the link with "Authentication failure" error code

**Impact**
Failing GAP/PRIV/CONN/BI-01-C qualification test

**Screenshots or console output**
[gap_priv_conn_bi-01-c.log](https://github.com/zephyrproject-rtos/zephyr/files/3472314/gap_priv_conn_bi-01-c.log)
[gap_priv_conn_bi-01-c.btsnoop.log](https://github.com/zephyrproject-rtos/zephyr/files/3472316/gap_priv_conn_bi-01-c.btsnoop.log)

**Environment (please complete the following information):**
 - nrf52840

**Additional context**
Tested with below patches applied:
882504af8bc2458eb63a9487e0f3b70d8b11aff8
dd11270ae0c3fd8c3ed67f567676f298e196db7c
3260d6f2481ec5046c82979a099d6af8f1a1bf2f

joerchan, 2019-08-14:
This seems to me to imply that the there can only be one bond for a peripheral device. So this might be from the time where there was only allowed to have one peripheral connection.
Also this test belongs in the privacy 1.0 category, we should run 1.2.x privacy test cases.

I see no reason to demand authentication with an unresolved peer, it's up to the application if authentication is required for connections. 

I suggest to close this with no action.

joerchan, 2019-08-15:
Closing because this is old privacy behavior.

closed 2019-08-15 by joerchan