zephyrproject-rtos/zephyr #18050: BT - Advertisement extensions API

aescolar, 2019-08-06:
Placeholder to agree on the advertisement extensions API in the host

kapi-no, 2019-09-25:
Could you also fix the problem of starting an advertiser when Host has run out of connection objects as part of this task?

carlescufi, 2019-09-26:
> Could you also fix the problem of starting an advertiser when Host has run out of connection objects as part of this task?

I think this is different @kapi-no. I suggest @joerchan takes a look at this separately.

joerchan, 2019-09-26:
@carlescufi We have discussed this on slack,and there is a change that we have decided to take as part of the AE work that would fix the issue mentioned. which is why @kapi-no added the reminder for it.

closed 2020-03-29 by jhedberg