zephyrproject-rtos/zephyr #18052: z_fatal_error missing log_strdup

jimparis, 2019-08-06:
I get an error about `log_strdup` on fatal errors:

```
[00:00:00.460,327] <err> os: ***** USAGE FAULT *****
[00:00:00.460,357] <err> os:   Division by zero
[00:00:00.460,357] <err> os: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x00000000
[00:00:00.460,418] <err> os: r3/a4:  0x00000001 r12/ip:  0x000397a7 r14/lr:  0x0000af4b
[00:00:00.460,418] <err> os:  xpsr:  0x01000000
[00:00:00.460,723] <err> os: Faulting instruction address (r15/pc): 0x0000af52
[00:00:00.460,754] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception
[00:00:00.460,784] <err> os: Current thread: 0x200058fc (main)
[00:00:01.409,942] <err> log: argument 1 in log message "Current thread: %p (%s)" missing log_strdup().
[00:00:01.528,259] <err> os: Halting system
```
Which is just an annoyance, but if `CONFIG_ASSERT` were set, this would trigger an assertion failure too.

andrewboie, 2019-08-06:
@nordic-krch for a variety of reasons I would really like to avoid making heap memory allocations in the kernel's fatal error handling code. Are there any alternatives?

andrewboie, 2019-08-06:
Further detail on this problem:

- Any use of z_fatal_print() with a string will result in this problem. This could be anywhere in arch code, not just kernel/fatal.c
- z_fatal_print() is *supposed* to be workable with either a printk() or the logging subsystem. We can't just put log_strdup() calls in invocations since this might be printk backed
- we have no need for fatal error handling printout to be optimized, as these are all pathological situations

Given all that, I think the best thing to do is have z_fatal_print() render the string at the call site and pass that to the log mechanism

nordic-krch, 2019-08-07:
> @nordic-krch for a variety of reasons I would really like to avoid making heap memory allocations in the kernel's fatal error handling code. Are there any alternatives?

@andrewboie I've prepared #18076. Imo, it is good to keep logger in the fatal error handler. At some point we will definitely see flash log backend (crashlog) and then having log in flash will become very handy. Even now, someone can do that out of tree and by having logs there it's possible.

PR #18060 (2019-08-07): kernel: temp remove log system support for fatal msgs
This needs further design work due to problems with logging
C strings. Just send always to printk() for now until this
is resolved.

Fixes: #18052
commit b6d961b7 (2019-08-07): kernel: remove log system support for fatal msgs

This needs further design work due to problems with logging
C strings. Just send always to printk() for now until this
is resolved.

Fixes: #18052
closed 2019-08-07 by andrewboie