zephyrproject-rtos/zephyr #18080: LWM2M bootstrap issue

rlubos, 2019-08-07:
I'm not sure if the issue I describe here is a bug report or my misunderstanding, hence not marking it as a bug yet. The documentation is quite scarce on how to initiate the bootstrap with the LWM2M library so I might miss some information.

I've tried to run the LWM2M bootstrap in the `lwm2m_client` sample, using local LWM2M bootstrap server (`leshan-bsserver-demo.jar` from the same source as the `leshan-server-demo.jar` from docs). What I did (all w/o security enabled):
1.  Started the local BS server instance at port 5683, along with local LWM2M server instance at port 5783.
2. Configured the BS server with client endpoint name (`qemu_x86`) and local LWM2M server address (`coap://localhost:5783`)
3. Enabled bootstrap support in the sample (`CONFIG_LWM2M_RD_CLIENT_SUPPORT_BOOTSTRAP`) and increased the maximum number of security and server object instances.
4. Configured the security object, to indicate it points to a bootstrap server (`lwm2m_engine_set_u8("0/0/1", 1);`)

After this, I was able to initiate the LWM2M bootstrap procedure, yet the client was not able to finalize it.  It seems, that the procedure that should set the appropriate state in the state machine (`engine_bootstrap_finish()`) is not called anywhere.  When the server sent the Bootstrap-finish message, the client replied with 4.04 Not Found. This clearly indicates that I miss a handler for the `Bootstrap-finish` message. 

[image]

So my question is, is the the `Bootstrap-finish` handler supposed to be present in LWM2M engine and we miss it or is the application supposed to provide it and register it somehow with LWM2M on its own?

Also, I have a side question regarding client deregistration - is it possible to deregister a LWM2M server with current impelmentation? I saw there is a `ENGINE_DEREGISTER` which seems to handle the deregistration procedure, yet I've seen no way to set this state. Is it possibe to trigger deregistration from the application somehow?

mike-scott, 2019-08-07:
@rlubos It's probably a bug :)  I will run some tests today.

mike-scott, 2019-08-16:
So by "today" I meant "9 days from now".  Looking at this now.

closed 2019-08-24 by ioannisg