zephyrproject-rtos/zephyr #18090: [zephyr 1.14] Mesh Friend queues more messages than indicates it's Friend Cache

MariuszSkamra, 2019-08-07:
**Describe the bug**
This is causing MESH/NODE/FRND/FN/BV-08-C qualification test case fail.
Zephyr Friend Cache Size is set to 16.
In the first step, PTS_1 sends 14 messages to Zephyr, that are queued and later polled by PTS_2.
In the second step, PTS_1 sends 19 messages to Zephyr. PTS_2 polls the messages and receives 19 packets instead of 15 or 16 as PTS states. 

According to Mesh Profile 1.0.1 3.5.5 Friend Queue
```
If the Friend Queue is full and a new message needs to be stored that is not a Friend Update message,
the oldest entries other than a Friend Update message shall be discarded to make room for the new
message.
Note: An implementation may have to discard multiple messages to fit the new message into the
Friend Queue.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run MESH/NODE/FRND/FN/BV-08-C test case

**Impact**
MESH/NODE/FRND/FN/BV-08-C

**Screenshots or console output**
[mesh_node_frnd_fv_bv-08-c.log](https://github.com/zephyrproject-rtos/zephyr/files/3477320/mesh_node_frnd_fv_bv-08-c.log)

jhedberg, 2019-08-14:
Ok, this is kind of annoying and something I'd consider raising an errata. The implementation does reserve some extra buffers, and reserves this amount for as many LPNs as it's configured to support. This means that it will be able often queue more than it promised. When ultimately the buffers run out there's a mechanism to go discard buffers from the existing queues, ensuring that we can always store at least the amount that we promised. So I wouldn't really consider this a bug, assuming I understood the issue correctly.

jhedberg, 2019-08-14:
Anyway, I can fix this fairly easily by removing the buffer discarding. The code is already keeping track of how many messages are in each queue. However, the more reasonable test spec requirement would be to test that the Friend can queue as many messages as it promises, but is allowed to queue more if it's able to.

jhedberg, 2019-08-14:
@trond-snekvik FYI

jhedberg, 2019-08-14:
@MariuszSkamra what's the value of `CONFIG_BT_MESH_FRIEND_LPN_COUNT` for the tester you're using? It seems it's more than one since otherwise the queue could contain only one extra message:
```
/* We reserve one extra buffer for each friendship, since we need to be able    
 * to resend the last sent PDU, which sits separately outside of the queue.     
 */                                                                             
#define FRIEND_BUF_COUNT    ((CONFIG_BT_MESH_FRIEND_QUEUE_SIZE + 1) * \         
                             CONFIG_BT_MESH_FRIEND_LPN_COUNT)    
```

jhedberg, 2019-08-15:
>  I can fix this fairly easily by removing the buffer discarding and adding a strict limit per queue. The code is already keeping track of how many messages are in each queue.

Actually it's not so simple due to the need to handle segmented messages. Those are stored in a separate list prior to insertion into the friend queue, yet they get allocated from the same buffer pool. I'm working on a fix, but it'll be slightly more complex (e.g. requiring some refactoring of the interface between transport.c and friend.c).

MariuszSkamra, 2019-08-16:
@jhedberg ```CONFIG_BT_MESH_FRIEND_LPN_COUNT=2```

PR #18479 (2019-08-19): Bluetooth: Mesh: Fix adhering to the configured Friend Queue size
Qualification test case MESH/NODE/FRND/FN/BV-08-C requires that we do
not store more messages than the reported Friend Queue size. The
implementation was so far opportunistic and stored more if it could
(it would later discard if necessary to make sure all queues can store
the required amount). The spec also requires the queues to have new
messages overwrite old ones (in the style of a circular buffer), so we
have to keep track of which buffers are part of the same segmented
message (so we discard all buffers belonging to the sa [... cut]
commit 8aba96b7 (2019-08-19): Bluetooth: Mesh: Fix adhering to the configured Friend Queue size

Qualification test case MESH/NODE/FRND/FN/BV-08-C requires that we do
not store more messages than the reported Friend Queue size. The
implementation was so far opportunistic and stored more if it could
(it would later discard if necessary to make sure all queues can store
the required amount). The spec also requires the queues to have new
messages overwrite old ones (in the style of a circular buffer), so we
have to keep track of which buffers are part of the same segmented
message (so we discard all buffers belonging to the s [... cut]
commit ce83a7c0 (2019-08-20): Bluetooth: Mesh: Fix adhering to the configured Friend Queue size

Qualification test case MESH/NODE/FRND/FN/BV-08-C requires that we do
not store more messages than the reported Friend Queue size. The
implementation was so far opportunistic and stored more if it could
(it would later discard if necessary to make sure all queues can store
the required amount). The spec also requires the queues to have new
messages overwrite old ones (in the style of a circular buffer), so we
have to keep track of which buffers are part of the same segmented
message (so we discard all buffers belonging to the s [... cut]
closed 2019-08-19 by jhedberg