zephyrproject-rtos/zephyr #18118: samples/subsys/console doesn't work with qemu_riscv32

nmeum, 2019-08-08:
**Bug description**

The sample application `samples/subsys/console` doesn't work with `qemu_riscv32`. This is due to the fact that the sifive uart [implemented by qemu](https://github.com/qemu/qemu/blob/b98a66201dbc7cf3b962f4bb260f66100cc75578/hw/riscv/sifive_uart.c) doesn't raise TX interrupts if `txcnt` is zero. From the [E310-G000 Manual](https://sifive.cdn.prismic.io/sifive%2F4d063bf8-3ae6-4db6-9843-ee9076ebadf7_fe310-g000.pdf) (Section 19.7):

```
The txwm condition becomes raised when the number of entries in the transmit FIFO is strictly less than the count specified by the txcnt field of the txctrl register.
```

The issues should also be reproducible with the `hifive1`, as it uses the same UART. Unfortunately, I currently don't have access to one and therefore couldn't test this.

Setting `UART_SIFIVE_PORT_0_TXCNT_IRQ` to a value `> 0` fixes this issue . I am therefore not sure if you would consider this a bug, I just wanted to document this somewhere since it is very confusing to be told that `the (interrupt-driven) console device doesn't work as expected` with the default configuration.

**To Reproduce**

In `samples/subsys/console`:

1. `mkdir build; cd build`
2. `cmake -GNinja -DBOARD=qemu_riscv32 ..`
3. `ninja run`

**Expected behavior**

Input should be echoed back.

**Actual behaviour**

```
***** Booting Zephyr OS build zephyr-v1.14.0-3004-g5ef7eea6e241 *****
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
```

No input is echoed back.

**Environment**

 - OS: Alpine Linux Edge
 - Commit SHA: 5ef7eea6e2411e126d2b764093e2a4d7bd380619

galak, 2019-08-13:
Can you try this again with the latest SDK (0.10.2) we pulled in a number of QEMU RISC-V fixes and the 'echo' test appears to work for me with it.

nmeum, 2019-08-18:
I didn't use the zephyr SDK, I used vanilla qemu 4.0.0, which version is shipped by the SDK? I managed to get hold of a hifive 1 board though, will test this application on it soonish.

nmeum, 2019-08-28:
> Can you try this again with the latest SDK (0.10.2) we pulled in a number of QEMU RISC-V fixes and the 'echo' test appears to work for me with it.

Just installed `zephyr-sdk-0.10.3` and tried `samples/subsys/console/echo`, same issue:

```
zephyr/samples/subsys/console/echo$ git describe
zephyr/samples/subsys/console/echo$ west build -b qemu_riscv32
…
zephyr/samples/subsys/console/echo$ west build -t run
***** Booting Zephyr OS build v2.0.0-rc1-274-g7c2f0636256d *****
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
```

Also keep in mind what the specification says about `txcnt == 0`, the behaviour qemu is showing is what I would expect.

> I managed to get hold of a hifive 1 board though, will test this application on it soonish.

I tried doing that but flashing doesn't seem to work:

```
zephyr/samples/subsys/console/echo$ west build -b hifive1
zephyr/samples/subsys/console/echo$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner hifive1
Open On-Chip Debugger 0.10.0+dev-00992-g3333261df-dirty (2019-08-16-00:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive, Inc.), part: 0x0e31, ver: 0x1)
Info : Examined RISCV core; XLEN=32, misa=0x40001105
Info : Listening on port 3333 for gdb connections
Info : [0] Found 2 triggers
halted at 0x204007a0 due to debug interrupt
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Error: The 'riscv.cpu configure' command must be used before 'init'.

Remote communication error.  Target disconnected.: Connection reset by peer.
You can't do that when your target is `exec'
```

Is that a known issue?

I did flash the application using the OpenOCD configuration from [RIOT](https://github.com/RIOT-OS/RIOT). That worked but unfortunately I didn't get any output on the TTY. This doesn't seem to be an issue with the flashing setup though, since flashing the zephyr hello_world application works with the correct output on the TTY.

pabigot, 2020-03-24:
I'm going to assume this has been fixed by #23699, which fixes it for real hardware.  If not please re-open.`

closed 2020-03-24 by pabigot