zephyrproject-rtos/zephyr #18124: synchronization example fails to build for SMP platforms

abrodkin, 2019-08-08:
Initially reported by Shippable here https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/47829/3/console while reviewing https://github.com/zephyrproject-rtos/zephyr/pull/17992.

Yet could be reproduced for any other SMP platform, for example `qemu_x86_64`:
```
# mkdir build
# cd build
# cmake -DBOARD=qemu_x86_64 ../samples/synchronization/
```
Enable OpenOCD support (`CONFIG_OPENOCD_SUPPORT=y`).

```
# make
...
[ 60%] Built target app
In file included from .../include/kernel_includes.h:16,
                 from .../include/kernel.h:17,
                 from .../kernel/include/kernel_structs.h:10,
                 from .../subsys/debug/openocd.c:7:
.../subsys/debug/openocd.c:39:35: error: 'struct z_kernel' has no member named 'current'
  [OPENOCD_OFFSET_K_CURR_THREAD] = offsetof(struct z_kernel, current),
                                   ^~~~~~~~
.../subsys/debug/openocd.c:66:2: warning: #warning Please define OPENOCD_OFFSET_T_STACK_PTR for this architecture [-Wcpp]
 #warning Please define OPENOCD_OFFSET_T_STACK_PTR for this architecture
  ^~~~~~~

```

ioannisg, 2019-08-20:
@abrodkin is this issue been taken care off?

abrodkin, 2019-08-20:
@ioannisg sorry I was on ETO last 2 weeks with limited access to Internet, now I'm going to re-spin https://github.com/zephyrproject-rtos/zephyr/pull/18125 with fix in for of `!SMP` dependency of `OPENOCD_SUPPORT` in subsys/debug/Kconfig. Is that OK?

galak, 2019-08-20:
there are 2 issues here, one is the SMP support, the other is building on X86_64.

So I think #18125 deals with the SMP issue.

@andrewboie does non-SMP on x86_64 make sense?

abrodkin, 2019-08-20:
@galak what's the problem with building on X86_64? Maybe I missed some discussion last weeks so pls kindly point me to any relevant information and I'll take a look.

galak, 2019-08-20:
> @galak what's the problem with building on X86_64? Maybe I missed some discussion last weeks so pls kindly point me to any relevant information and I'll take a look.

The warning `#warning Please define OPENOCD_OFFSET_T_STACK_PTR for this architecture` is related to x86_64

abrodkin, 2019-08-20:
> 
> 
> > @galak what's the problem with building on X86_64? Maybe I missed some discussion last weeks so pls kindly point me to any relevant information and I'll take a look.
> 
> The warning `#warning Please define OPENOCD_OFFSET_T_STACK_PTR for this architecture` is related to x86_64

Hm... but how can I reproduce it?
I don't see it with:
```
cmake -DBOARD=qemu_x86_64 ../samples/synchronization/
make
```

abrodkin, 2019-08-20:
> Hm... but how can I reproduce it?
> I don't see it with:
>
> ```
> cmake -DBOARD=qemu_x86_64 ../samples/synchronization/
> make
> ```

Ooops, forgot to enable `CONFIG_OPENOCD` via menuconfig :(
Now I see it but again this has nothing to do with anything except `x86_64` and `CONFIG_OPENOCD`.

So it looks like we'd need this then (since `X86_64` seems to be quite incomplete, judging by contents of https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/x86_64/include/kernel_arch_thread.h):
```
diff --git a/subsys/debug/Kconfig b/subsys/debug/Kconfig
index a5dff0ad4f..202d7f3f29 100644
--- a/subsys/debug/Kconfig
+++ b/subsys/debug/Kconfig
@@ -223,7 +223,7 @@ config EXCEPTION_STACK_TRACE

 config OPENOCD_SUPPORT
        bool "OpenOCD support [EXPERIMENTAL]"
-       depends on !SMP
+       depends on !SMP && !X86_64
        select THREAD_MONITOR
        select THREAD_NAME
        help
```

Does that look OK?

closed 2020-05-10 by carlescufi