zephyrproject-rtos/zephyr #18131: devicetree should check input against declared type

pabigot, 2019-08-08:
**Is your enhancement proposal related to a problem? Please describe.**

Users are assuming that a property `jedec-id` that has type `uint8-array` can accept a value as `< 0xc2 0x28 0x17>`, but that will generate a 12-byte initializer for three `u32_t` values when `[c8 28 17]` would generate the correct expected 3-byte initializer.

**Describe the solution you'd like**

The expected type of values is provided in the binding.  It should be checked.

**Describe alternatives you've considered**

Allowing users to have to figure it out by themselves, or ask on slack.

aescolar, 2019-08-08:
CC @galak @ulfalizer

ulfalizer, 2019-08-08:
This is trickier than it might seem at a glance (think `foo = <1 2 3>, "bar", [00 01], &hmm;`), so it'll take some work, but I'll get on it.

Gonna start with marking up offsets where different data blocks start (already do that for e.g. phandles). Then a "type" could be derived from that later.

closed 2019-08-13 by galak