zephyrproject-rtos/zephyr #18143: stm32f SPI TX does not work correctly.

likeaferrari, 2019-08-09:
- Environment
ubuntu 18.04
Zephyr 1.14.99
STM32F410RB (nucleo F410RB)

I'm working  on the STM32F410RB. When transmit 8bytes data in SPI Slave mode, it occurs Overrun error and does not send correctly.

 I tested with simple send code as like below.

```
#include <errno.h>
#include <zephyr.h>
#include <misc/printk.h>
#include <device.h>
#include <spi.h>
#include "stm32f4xx_hal.h"

static int koo_spi_write(struct device *spi, struct spi_config *spi_cfg,
			    u8_t cmd, u8_t addr, u8_t data)
{
	char tmp[8] = { 1,2,3,4,5,6,7,8};
	struct spi_buf bufs = {
			.buf = tmp,
			.len = 8
	};
	struct spi_buf_set tx = {
		.buffers = &bufs
	};
	
	tx.count = 1;

	return spi_write(spi, spi_cfg, &tx);
}

void main(void)
{
	struct device *spi;
	struct spi_config spi_cfg;
	struct spi_cs_control spi_cs;
	int err, i;
	char data[8];

	printk("KOO SPI example application\n");

	spi = device_get_binding(DT_SPI_1_NAME);
	if (!spi) {
		printk("Could not find SPI driver\n");
		return;
	}

	spi_cfg.operation = SPI_WORD_SET(8) | SPI_OP_MODE_SLAVE;
        spi_cfg.frequency = 25000000;

	spi_cfg.cs = NULL;
	err = koo_spi_write(spi, &spi_cfg, (0xff- i), i, 0x21);

}
```

and in the recv side, test message show the result like this.

> ...
> recv: 01 02 02 02 02 02 02 02
> recv: 01 02 02 02 02 02 02 02
> recv: 01 02 02 02 02 02 02 02
> recv: 01 02 02 02 02 02 02 02
> ...

I tracked the flow, and found LL_SPI_IsActiveFlag_TXE(spi) does not return true cause TX->SR does not indicate TX Empty. as a result, TX frame cannot be shifted and MASTER takes duplicated data.

```
zephyr/dirvers/spi/spi_ll_stm32.c 
static void spi_stm32_shift_s(SPI_TypeDef *spi, struct spi_stm32_data *data)
{
	if (LL_SPI_IsActiveFlag_TXE(spi) && spi_context_tx_on(&data->ctx)) {
		u16_t tx_frame = spi_stm32_next_tx(data);

		if (SPI_WORD_SIZE_GET(data->ctx.config->operation) == 8) {
			LL_SPI_TransmitData8(spi, tx_frame);
			spi_context_update_tx(&data->ctx, 1, 1);
		} else {
			LL_SPI_TransmitData16(spi, tx_frame);
			spi_context_update_tx(&data->ctx, 2, 1);
		}
```
And finally SR register shows OVERRUN flag.

At first time I thought there're problem with my source or configurations,
But when I perform same work on MASTER mode, TX/RX both all right,
and I also tried same work with STM HAL API and it works very well too.

I'm not sure but I doubt Zephyr has problem handling SR.

erwango, 2019-08-13:
@likeaferrari, can you provide the following, so I can have a try on my end in same conditions?
- code you're using on master side
- SHA1 (code changes quickly these days)
- pin settings

b0661, 2019-08-13:
You are transmitting at 25 MHz.
```
spi_cfg.frequency = 25000000;
```

Are you shure your system is fast enough to handle that? Try with a lower frequency. It looks like the Tx data reload is to slow.

likeaferrari, 2019-08-14:
@erwango Hello. I cannot access my develop environment until next Monday, but I'll tell you requested things as soon as i come back. 

@b0661 That was what I focused at first time.  I tested almost every 1Mhz of both side. of course there was some differences when frequency changed, but I could not reach to ideal result.

b0661, 2019-08-14:
> That was what I focused at first time. I tested almost every 1Mhz of both side. of course there was some differences when frequency changed, but I could not reach to ideal result

I was running SPI in interrupt mode and had interferences of other interrupts (e.g. logging on RS232) resulting in data loss. I had to switch down to 187500 Hz (the lowest possible frequency for my configuration) to get a long time reliable data exchange in my test configuration (Raspi, Linux, SPI master -> Nucleo-F091RC, Zephyr, SPI slave). For byte transfers this resulted in an interrupt every ~43usec. Running the transfer at 25MHz as you do would result in an interrupt every ~0.3usec. In the end the project decided to switch to custom made DMA mode operation which I think is a far better solution if you want to go to such a high transfer frequency.

likeaferrari, 2019-08-19:
@erwango 
Here's the info what you requested.
SHA: bd001081fce1e2b2123f8248be9c3c938d1f470c

I made test environment as like below picture.
[image]

and here is my part of pinmux.c indicate SPI1 configs.
[image]

and this is test source of master and slave. (gist link)
https://gist.github.com/likeaferrari/69f253477996d21757cc357372542c5e

also I checked menuconfig each case of master and slave. (this is config of slave side)
[image]

Finally, when I run both board and DEBUG show the msgs like below. (left: M / right: S)
[image]

Although sending from MASTER do not work perfectly, reply from SLAVE never successes.
This is all what I tested.
and as I mentioned before, sending from SLAVE fails because LL_SPI_IsActiveFlag_TXE(spi) returns OVERRUN flag.

but when try same work with STM32 HAL spi API, It works very well though it has process similar with zephyr spi functions.

erwango, 2019-08-28:
@likeaferrari, thanks for these inputs.
Can you also share you dts configuration? I'll try to reproduce but, some details might differ and details are where devil hides. Maybe sharing a branch would be the fastest.
Anyway, I'll have some tries with information provided and will keep you posted.

erwango, 2019-08-30:
@likeaferrari, when using koo_spi_write_then_read instead of koo_spi_write/ koo_spi_read (functions extracted from your code in gist, I get no overrun errors, and Tx/Rx happens with no byte loss.
Not looked deeply into it, but I guess this is normal given the samples code proposed.
Btw, tx issue is not directly linked to OVR, since OVR is a data read issue. Though I think that both symptoms come from the same issue.

Please check the following code:
https://gist.github.com/erwango/4165f917e75e72ef49fd976c5d57d0ef

With output at master's side:
```
Sent: 1122, received :5566                                                   
Sent: 3344, received :1122                                                   
Sent: 5566, received :3344                                                   
Sent: 1122, received :5566                                                   
Sent: 3344, received :1122                                                   
Sent: 5566, received :3344                                                   
Sent: 1122, received :5566
```

erwango, 2019-11-01:
@likeaferrari, is the issue still active?

closed 2019-11-04 by likeaferrari