zephyrproject-rtos/zephyr #18154: Qemu: mps2+: missing documentation

ioannisg, 2019-08-09:
There needs to be some documentation under
-  boards/boards/arm/mps2_an385/doc
-  boards/boards/arm/mps2_an521/doc

That documentation should indicate that these are Qemu targets for ARM; how they can be used; known issues, etc.

Probably just copy-paste with trivial adjustments from some other board's QEMU documentation

See, for examples, qemu_cortex_m3

microbuilder, 2019-08-09:
@ioannisg These shouldn't be exclusively **qemu** boards, though. There is a PR here that I used to do TF-M dev and testing on the actual MPS2+ board with AN521, which is the board many people are using at ARM as well: https://github.com/zephyrproject-rtos/zephyr/pull/16450/files

I can update this today with a doc PR, but describing them as qemu only is overly limited in my opinion?

ioannisg, 2019-08-09:
No, I did not want them to be shown as Qemu-only boards; instead, there should be some information that we support running tests in CI (on Qemu) for these boards. Besides them being regular platforms. Sorry for the confusion.

Btw, doc/bug-fix PRs can be merged even after the feature freeze, so no particular rush on this

ioannisg, 2019-09-03:
@microbuilder would you be able to address this for v2.0.0?

PR #18908 (2019-09-05): arm: mps: docs: note on qemu usage of boards
This commit adds a note alluding to the fact that these two boards are
primarily included for use with QEMU, but have been tested on actual
hardware as well where appropriate.

Resolves #18154 .
closed 2019-09-05 by ioannisg