zephyrproject-rtos/zephyr #18155: i2c_ll_stm32_v1: I2C_CR1:STOP is not cleared

KwonTae-young, 2019-08-09:
**Describe the bug**
I am using STM32F415RG with I2C_2 (OLED), PWM (Buzzer), ADC, USART_1 (IrDA), SPI_1 (M-RAM).
There is a phenomenon that data of I2C is not output properly in random situation.
In this case, if i check the register of I2C, **`I2C_CR1:STOP`** bit is set.(0x00000200)
```
(gdb) x /10x 0x40005800
0x40005800:	0x00000200	0x0000002a	0x00000000	0x00000000
0x40005810:	0x00000078	0x00000000	0x00000000	0x00008023
0x40005820:	0x0000000d	0x00000000
```

In this case, the address of i2c_slave is not output when using [i2c_write()](https://docs.zephyrproject.org/1.14.0/reference/peripherals/i2c.html#_CPPv39i2c_writeP6devicePK4u8_t5u32_t5u16_t).
So only `START Condtion` and` STOP Condtion` are output.
From then on, this output is repeated.
The **`I2C_CR1:STOP`** bit is not cleared and remains set.
[image]

I look at the I2C_DR register, it contains 0x78, which is the slave address of SSD1306, 0x3C.
In my case OLED does not work because I2C write fails.

**To Reproduce**
I haven't found a definite condition to reproduce yet.
Currently, symptoms occur randomly.(< 10minutes)

**Expected behavior**
I2C should be written as usual.

**Impact**
In this case, no data can be written to the corresponding I2C channel.
Only `START Condition` and` STOP Condtion` are displayed.

**Screenshots or console output**
gdb step I2C waveform.
```
(gdb) continue
Continuing.
```
[image]
```
^C
Program received signal SIGINT, Interrupt.
k_cpu_idle () at /work/zephyrproject/zephyr/arch/arm/core/cpu_idle.S:139
139		bx lr
(gdb) x /10x 0x40005800
0x40005800:	0x00000000	0x0000002a	0x00000000	0x00000000
0x40005810:	0x00000000	0x00000000	0x00000000	0x00008023
0x40005820:	0x0000000d	0x00000000
(gdb) continue
Continuing.
```
[image]  
[image]  
[image]
```
^C
Program received signal SIGINT, Interrupt.
k_cpu_idle () at /work/zephyrproject/zephyr/arch/arm/core/cpu_idle.S:139
139		bx lr
(gdb) x /10x 0x40005800
0x40005800:	0x00000200	0x0000002a	0x00000000	0x00000000
0x40005810:	0x00000078	0x00000000	0x00000000	0x00008023
0x40005820:	0x0000000d	0x00000000
(gdb) set {int}0x40005800 = 0
(gdb) x /10x 0x40005800
0x40005800:	0x00000000	0x0000002a	0x00000000	0x00000000
0x40005810:	0x00000078	0x00000000	0x00000000	0x00008023
0x40005820:	0x0000000d	0x00000000
(gdb) continue
Continuing.
```
[image]

**Environment (please complete the following information):**
- OS: Ubuntu 18.04
- Toolchain (e.g Zephyr SDK, ...)
- Commit SHA: 6a01db3be062be27454e98dd29497099ead9feb5

**Additional context**
What I don't understand in this state is that `STOP Condtion` is output, but **`I2C_CR1: STOP`** is not cleared.

In this situation, clear `I2C_CR1` to work normally.
```
(gdb) x /10x 0x40005800
0x40005800:	0x00000200	0x0000002a	0x00000000	0x00000000
0x40005810:	0x00000078	0x00000000	0x00000000	0x00008023
0x40005820:	0x0000000d	0x00000000
(gdb) set {int}0x40005800 = 0
```

**`I2C_CR1:STOP`** needs to be checked but there are currently no LL driver functions to check.
So I tested with the following modifications.
This change has confirmed that I2C can recover from a problem.
``` diff
--- drivers/i2c/i2c_ll_stm32_v1.c_ori	2019-08-09 21:06:59.036586163 +0900
+++ drivers/i2c/i2c_ll_stm32_v1.c	2019-08-09 21:04:57.184586163 +0900
@@ -326,6 +326,11 @@

ARG_UNUSED(next_msg_flags);

+	if (READ_BIT(i2c->CR1, I2C_CR1_STOP)) {
+		CLEAR_BIT(i2c->CR1, I2C_CR1_STOP);
+		return -EIO;
+	}
+
LL_I2C_AcknowledgeNextData(i2c, LL_I2C_ACK);

if (msg->flags & I2C_MSG_RESTART) {
````

I'm sorry I do not have enough English. :)
Thank you.

ydamigos, 2019-08-12:
Hi @KwonTae-young,

could you please test with the latest master to check if the issue still exists?

KwonTae-young, 2019-08-13:
@ydamigos
I will test and share the results as soon as possible.

KwonTae-young, 2019-08-16:
@ydamigos 
Some problems occurred when merging master's source to my current working source.
I will reopen this issue once the simple sample code is ready.

closed 2019-08-16 by KwonTae-young