zephyrproject-rtos/zephyr #18157: adding an offset to the zephyr code via dts overlay breaks sys_reboot, log_init and z_impl_uart_line_ctrl_get + the image size changes

StefJar, 2019-08-09:
I am doing a change in the dts flash settings via an overlay in a shield.

my overlay file looks like(my.overlay):
```
&flash0 {
	partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        slot0_partition: partition@20000 {
                label = "zephyr-code";
                reg = <0x08020000 DT_SIZE_K(384)>;
        };
	};
};
```
Compiling without my.overlay:
```
scanning dependencies of target zephyr_prebuilt
[ 95%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 96%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      121012 B         1 MB     11.54%
            SRAM:       49588 B       256 KB     18.92%
        IDT_LIST:         568 B         2 KB     27.73%
[ 96%] Built target zephyr_prebuilt
Scanning dependencies of target linker_pass_final_script_target
[ 97%] Generating linker_pass_final.cmd
[ 97%] Built target linker_pass_final_script_target
[ 98%] Generating isr_tables.c
Scanning dependencies of target zephyr_final
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[100%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[100%] Linking C executable zephyr.elf
Generating files from zephyr.elf for board: myTestBoard
[100%] Built target zephyr_final
```
compiling with overlay:
```
Scanning dependencies of target zephyr_prebuilt
[ 96%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 97%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       68828 B         1 MB      6.56%
            SRAM:       47088 B       256 KB     17.96%
        IDT_LIST:         568 B         2 KB     27.73%
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(myUsbACM.c.obj): in function `uart_line_ctrl_get':
/home/stefan/git/test/build/zephyr/include/generated/syscalls/uart.h:41: undefined reference to `z_impl_uart_line_ctrl_get'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(myUsbACM.c.obj): in function `uart_line_ctrl_set':
/home/stefan/git/test/build/zephyr/include/generated/syscalls/uart.h:39: undefined reference to `z_impl_uart_line_ctrl_set'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(myUsbACM.c.obj): in function `uart_line_ctrl_get':
/home/stefan/git/test/build/zephyr/include/generated/syscalls/uart.h:41: undefined reference to `z_impl_uart_line_ctrl_get'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/stefan/git/test/targets/withOffs/build/zephyr/include/generated/syscalls/uart.h:41: undefined reference to `z_impl_uart_line_ctrl_get'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(cep.c.obj): in function `cep_add_rtml_fw_stm32_chunk':
/home/stefan/git/test/src/cep/cep.c:243: undefined reference to `sys_reboot'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(main.c.obj): in function `main':
/home/stefan/git/test/src/main.c:203: undefined reference to `log_init'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/stefan/git/test/src/main.c:245: undefined reference to `sys_reboot'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:103: zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:619: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
done!
```

andrewboie, 2019-08-09:
z_impl_uart_line_ctrl_get is in drivers/serial/uart_handlers.c

It is ifdefed around CONFIG_UART_LINE_CTRL.

Is CONFIG_UART_LINE_CTRL enabled in your Kconfig?
Is that C file being built?

StefJar, 2019-08-10:
@andrewboie guess so. I didn't changed code or config - only I used a user defined shield to insert a offset to the zephyr partition.
I checked the ".config" file in the cmake build dir. There is an entry for  `CONFIG_UART_LINE_CTRL=y`

Think this error is more releated to some internal linking stuff. 
if I am setting
`FLASH_LOAD_OFFSET = 0x20000`
The linking works

galak, 2019-08-13:
Can you try and reproduce with nucleo_f412zg.

I tried adding:

```
diff --git a/boards/arm/nucleo_f412zg/nucleo_f412zg.dts b/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
index e5f184f73a..8fc8d1001b 100644
--- a/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
+++ b/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
@@ -92,3 +92,16 @@ arduino_serial: &usart6 {};
 &rtc {
        status = "okay";
 };
+
+&flash0 {
+       partitions {
+               compatible = "fixed-partitions";
+               #address-cells = <1>;
+               #size-cells = <1>;
+
+               slot0_partition: partition@20000 {
+                       label = "zephyr-code";
+                       reg = <0x08020000 DT_SIZE_K(384)>;
+               };
+       };
+};
```

but didn't see any issue if I built hello_world sample

StefJar, 2019-08-18:
@galak putting it into the original dts works. no problem
Now try to create an overlay and put your patch into it. Remove the flash0 entry from the dts file. Add the overlay via "shield" to your project and build. Think you will see a linker fail.
Now add to your project file `FLASH_LOAD_OFFSET=0x20000` and build again. This time the linking should work.

galak, 2019-08-20:
> @galak putting it into the original dts works. no problem
> Now try to create an overlay and put your patch into it. Remove the flash0 entry from the dts file. Add the overlay via "shield" to your [...]

Can you post something to github on a branch or something that would make it easier for me to reproduce what you are seeing.

galak, 2020-07-22:
is this still an issue?  Haven't seen a response in almost over a year - so going to close and please reopen if it is.

closed 2020-07-22 by galak