zephyrproject-rtos/zephyr #18160: Cleanup dts compatible for "nxp,kinetis-sim" on nxp_ke1xf

galak, 2019-08-09:
"nxp,kinetis-sim" is used in the following dtsi files:

```
dts/arm/nxp/nxp_k6x.dtsi:                       compatible = "nxp,kinetis-sim";
dts/arm/nxp/nxp_ke1xf.dtsi:                     compatible = "nxp,kinetis-sim";
dts/arm/nxp/nxp_kl25z.dtsi:                     compatible = "nxp,kinetis-sim";
dts/arm/nxp/nxp_kw2xd.dtsi:                     compatible = "nxp,kinetis-sim";
dts/arm/nxp/nxp_kw40z.dtsi:                     compatible = "nxp,kinetis-sim";
dts/arm/nxp/nxp_kw41z.dtsi:                     compatible = "nxp,kinetis-sim";
```

However nxp_ke1xf.dtsi is the only one what doesn't use it as a clock controller.  The HW is a little different and as such we should use a different compatible for it on nxp_ke1xf.

henrikbrixandersen, 2019-08-09:
@galak Could you please elaborate a bit on why a different `compatible` is needed? What problem does the current solution pose?

galak, 2019-08-09:
> @galak Could you please elaborate a bit on why a different `compatible` is needed? What problem does the current solution pose?

The idea with compatible is that it should be used if the software/programming model is the same for the HW.  While I'm guessing the manual's call the block SIM on all the Kinetis SoC's, it seems as if the way clocks and SIM work differ greatly between the KE1XF and other SoC's.  As such we should use a slightly different compatible on KE1XF.

Such as "nxp,kinetis-ke1xf-sim" and introduce a binding specific to it.

galak, 2019-08-09:
Got a patch in progress

PR #18162 (2019-09-11): dts: nxp: ke1xf: Introduce nxp,kinetis-ke1xf-sim
The KE1xF SoC family SIM unit differs from the other Kinetis family
SoCs.  Add a unique compatiable and binding for it.

Fixes #18160
commit 57b10d09 (2019-09-11): dts: nxp: ke1xf: Introduce nxp,kinetis-ke1xf-sim

The KE1xF SoC family SIM unit differs from the other Kinetis family
SoCs.  Add a unique compatiable and binding for it.

Fixes #18160
closed 2019-09-11 by galak