zephyrproject-rtos/zephyr #18169: dts: bindings: inconsistent file names and base.yaml include of general device controllers

b0661, 2019-08-10:
**Describe the bug**

Binding files for general device controllers (e.g. gpio controller, interrupt controller, spi controller) are named inconsistently (e.g. gpio-controller.yaml vs. spi.yaml).

Inclusion of base.yaml by the general device controller bindings is inconsistent. E.g. gpio-controller.yaml does not include base.yaml whereas spi.yaml includes base.yaml.

**To Reproduce**

See general bindings for device controllers.

**Expected behavior**

Consistent binding file naming. Remove "-controller" from gpio-controller.yaml and interrupt-controller.yaml.
Consistent include of base.yaml by the general device controller binding; Make all general device controller bindings include base.yaml. 

**Impact**

Annoyance

**Screenshots or console output**
N/A

**Environment (please complete the following information):**

Lates Zephyr

**Additional context**
N/A

ulfalizer, 2019-08-10:
`base.yaml` was added pretty recently by Galak. Just a bunch of boring batch replacement that hasn't been done yet.

If you have nothing better to do, you could submit some PRs for the ones you see, and I'll approve quickly. We should get to it later otherwise.

ulfalizer, 2019-08-12:
Personally, I'd add `-controller` where it's missing rather than remove it though. `!include interrupt-controller.yaml` is a lot clearer than `!include interrupt.yaml`.

b0661, 2019-08-12:
I think interrupt-controller is ok as the driver sub directory is also interrupt-controller. 

I looked at https://github.com/devicetree-org/dt-schema/tree/master/schemas to see how it is done for Linux but here it is not consistent either. 

What IMHO should be done is to keep 

- the Zephyr driver header file names (e.g. drivers/gpio/gpio.h) and
- the Zephyr bindings header file names (e.g. dt-bindings/gpio/gpio.h).and
- the Zephyr schema file names (e.g. bindings/gpio/gpio.yaml) and
- the Linux schema file names (e.g. schemas/gpio/gpio.yaml)

in sync as much as possible.

The pattern should be \<drivers | dt-bindings |  bindings | schemas\>/\<device\>/\<device\>.\<h | yaml\> for devices.

ulfalizer, 2019-12-04:
Has been fixed.

closed 2019-12-04 by ulfalizer