zephyrproject-rtos/zephyr #18173: ARM: Core Stack Improvements/Bug fixes for 2.1 release

ioannisg, 2019-08-11:
This is an umbrella issue for the ARM stack enhancements, including improvements/enhancements made for ARM SoCs and Boards, beyond 2.0 and towards the 2.1 release.

Several of the items below are  related to issues, which have been spotted before and during the 2.0 feature freeze and tree stabilization, and have, eventually, been moved to the 2.1 Zephyr release.

Items are constantly been added in the list, as issues are been reported and fixed. This will give a quick overview of the implemented fixes/improvements and help with the ARM release notes.

Areas:
- ARM stack 
- ARM SoCs, Boards, Drivers
- Kernel interface to arch
- Memory protection related tests

Scope
- Optimizations
- Coverage
- Feature enhancements
- Bug fixes / Workarounds (or workaround removals, if applicable)
- Clean-up, refactoring

Below is a list of tasks:

- [ ] [Bug ARM] Support for memory protection features in baseline cores (M0+ and M23) (#12961)
- [ ] [Bug ARM, Kernel] Make threads use `K_THREAD_STACK_SIZEOF `and remove workaround for ARM 's `z_new_thread()` (#14269)
- [ ] [Bug ARM] Support for HW stack protection features together with CONFIG_ARM_SECURE_FIRMWARE=y and CONFIG_FP_SHARING (currently broken)
- [ ] [Enh. ARCHitectures, Kernel] Support distinct ways for allocating stack areas for default and supervisor-only threads (#13637)
- [ ] [Enh. ARM] Add HW stack protection during boot, too.
- [ ] [Enh. ARMv8] Make built-in stack overflow a default feature (=y)
- [ ] [ARM, ARC, Kernel] Consider moving the arm_core_mpu API above the arch- level, so as not to duplicate for ARM, ARC
- [ ] [Enh. Kernel] Rename `userlocal_data `to something that does not contain "user" in the name, as it is used under `CONFIG_USERSPACE ` but beyond user threads
- [ ] [Enh. ARM] Document the incompatibilities between ARMv7-M and ARMv8-M Mainline

andrewboie, 2019-09-23:
@ioannisg please consider adding https://github.com/zephyrproject-rtos/zephyr/issues/19067 to the list.
I can take a look at implementing it if you don't have the bandwidth.

ioannisg, 2019-09-23:
> 
> 
> @ioannisg please consider adding #19067 to the list.
> I can take a look at implementing it if you don't have the bandwidth.

You're right  @andrewboie that one should be in for 2.1. I 'll include it.

ioannisg, 2019-12-05:
Unfinished issues moved to  #21201

closed 2019-12-05 by ioannisg