zephyrproject-rtos/zephyr #18183: [zephyr 1.14] GATT Server does not send inform change-unaware client about DB changes

MariuszSkamra, 2019-08-12:
**Describe the bug**
The client that was once informed about GATT DB change is not informed when DB changes once more between connections.

Core Spec 5.1 | Vol 3, Part G, 2.5.2.1 Robust Caching
```
The error response is sent only once after the client becomes change-unaware, unless the client disconnects or the database changes again before the client becomes change-aware in which case the error response shall be sent again.
```

**To Reproduce**
Execute GATT/SR/GAS/BV-07-C. Steps to reproduce the behavior:
0. PTS implementation of this test case is wrong, CASE0058236 bug report was created. Reproduce this using 2 Zephyr boards and TS steps.
1. Execute steps as in test case GATT/SR/GAS/BV-07-C
9. Zephyr fails to send "Database Out of Sync" error in step 9.

The problem is quite easy to reproduce with use of 2 boards flashed with bluetooth/shell:
1. CL: ```gatt read 0x000b```
```
> ACL Data RX: Handle 0 flags 0x02 dlen 7                                                                                                                                                                                                                                  #60 138.916200
      ATT: Read Request (0x0a) len 2
        Handle: 0x000b
< ACL Data TX: Handle 0 flags 0x00 dlen 11                                                                                                                                                                                                                                 #61 138.916500
      ATT: Read Response (0x0b) len 6
        Value: 68656c6c6f21
```
2. SR: ```gatt register```
3. CL: ```gatt read 0x000b```
```
> ACL Data RX: Handle 0 flags 0x02 dlen 7                                                                                                                                                                                                                                  #63 247.761100
      ATT: Read Request (0x0a) len 2
        Handle: 0x000b
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                                                  #64 247.761300
      ATT: Error Response (0x01) len 4
        Read Request (0x0a)
        Handle: 0x0000
        Error: Reserved (0x12)
```
4. SR: ```gatt unregister```
5. CL: ```gatt read 0x000b```
```
> ACL Data RX: Handle 0 flags 0x02 dlen 7                                                                                                                                                                                                                                  #66 336.125800
      ATT: Read Request (0x0a) len 2
        Handle: 0x000b
< ACL Data TX: Handle 0 flags 0x00 dlen 11                                                                                                                                                                                                                                 #67 336.126100
      ATT: Read Response (0x0b) len 6
        Value: 68656c6c6f21
```
As you can observe, Server returns the value of characteristic instead of error.

**Expected behavior**
Server returns "Database Out of Sync" error once per DB change.

**Impact**
Failing GATT/SR/GAS/BV-07-C conformance test case

**Screenshots or console output**
[gatt_sr_gas_bv_07_c.btsnoop.log](https://github.com/zephyrproject-rtos/zephyr/files/3491827/gatt_sr_gas_bv_07_c.btsnoop.log)

ioannisg, 2019-08-12:
Is this only a 1.14 bug, @MariuszSkamra ? Is it applicable for master?

MariuszSkamra, 2019-08-12:
@ioannisg I did not tested this against master. But it seems this could be applicable for master as well

Vudentz, 2019-08-15:
I am trying this out with BlueZ:

> gatt register 
[00:00:31.910,000] <dbg> bt_gatt.db_changed: <log_strdup alloc failed> change-unaware
[00:00:32.060,000] <dbg> bt_gatt.sc_indicate_rsp: b8:8a:60:d8:17:d7 (public) change-aware
> gatt unregister
[00:00:31.910,000] <dbg> bt_gatt.db_changed: <log_strdup alloc failed> change-unaware
[00:00:32.060,000] <dbg> bt_gatt.sc_indicate_rsp: b8:8a:60:d8:17:d7 (public) change-aware

Could it be perhaps that gatt unregister is failing for some reason, we are not really checking if bt_gatt_service_unregister succeeds or not? Anyway this is pretty strange as the gatt register/unregister do call into db_changed.

Vudentz, 2019-08-16:
So the culprint seems to be:

> ACL Data RX: Handle 0 flags 0x02 dlen 11                                                                                                                                           #133 [hci0] 103.342316
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x0001-0xffff
        Attribute type: Database Hash (0x2b2a)
< ACL Data TX: Handle 0 flags 0x00 dlen 24                                                                                                                                           #134 [hci0] 103.342616
      ATT: Read By Type Response (0x09) len 19
        Attribute data length: 18
        Attribute data list: 1 entry
        Handle: 0x0008
        Value: 74822c08a3f1c9a633be5dbaa577e880

So this will trigger:

	/* BLUETOOTH CORE SPECIFICATION Version 5.1 | Vol 3, Part G page 2347:
	 * 2.5.2.1 Robust Caching
	 * A connected client becomes change-aware when...
	 * The client reads the Database Hash characteristic and then the server
	 * receives another ATT request from the client.
	 */
	bt_gatt_change_aware(conn, true);

PR #18332 (2019-08-16): Bluetooth: GATT: Fix not clearing out of sync flag
When the database changes and a client becomes change unware it should
also clear the out of sync flag as the following request should return
an error:
```    
    Core Spec 5.1 | Vol 3, Part G, 2.5.2.1 Robust Caching:
    
      'The error response is sent only once after the client becomes
      change-unaware, unless the client disconnects or the database changes
      again before the client becomes change-aware in which case the error
      response shall be sent again.'
```
Fixes: #18183
commit b37efd35 (2019-08-16): Bluetooth: GATT: Fix not clearing out of sync flag

When the database changes and a client becomes change unware it should
also clear the out of sync flag as the following request should return
an error:

Core Spec 5.1 | Vol 3, Part G, 2.5.2.1 Robust Caching:

  'The error response is sent only once after the client becomes
  change-unaware, unless the client disconnects or the database changes
  again before the client becomes change-aware in which case the error
  response shall be sent again.'

Fixes: #18183
commit fc6a594a (2019-08-16): Bluetooth: GATT: Fix not clearing out of sync flag

When the database changes and a client becomes change unware it should
also clear the out of sync flag as the following request should return
an error:

Core Spec 5.1 | Vol 3, Part G, 2.5.2.1 Robust Caching:

  'The error response is sent only once after the client becomes
  change-unaware, unless the client disconnects or the database changes
  again before the client becomes change-aware in which case the error
  response shall be sent again.'

Fixes: #18183
commit 692d7d52 (2019-08-16): Bluetooth: GATT: Fix not clearing out of sync flag

When the database changes and a client becomes change unware it should
also clear the out of sync flag as the following request should return
an error:

Core Spec 5.1 | Vol 3, Part G, 2.5.2.1 Robust Caching:

  'The error response is sent only once after the client becomes
  change-unaware, unless the client disconnects or the database changes
  again before the client becomes change-aware in which case the error
  response shall be sent again.'

Fixes: #18183
closed 2019-08-16 by jhedberg