zephyrproject-rtos/zephyr #18188: [zephyr 1.14] Re-enabling CCC gets broken when used along with Robust Caching

MariuszSkamra, 2019-08-12:
**Describe the bug**
Zephyr Client that enabled Robust Caching and Subscribed for CCC Notifications may receive "ATT Out Of Sync" error while trying to re-enable indications after reconnection to bonded device. This breaks current implementation, because once Client receives an error on Write Request to CCC, it assumes it's unsubscribed.

**To Reproduce**
Steps to reproduce the behavior:
1. bond devices
2. CL: ```gatt subscribe 0x0004 0x0003 ind```(enable Service Changed indications)
3. CL: ```gatt write 0x0006 0 0x01``` (enable Robust Caching)
4. CL: ```bt disconnect```
5. SR: ```gatt register```
6. CL: ```bt connect dd:ad:fb:8f:b7:ae (random)```
7. Observe an "Unsubscribed" log

**Expected behavior**
Client shall handle ATT Out Of Sync error

**Screenshots or console output**
```
< ACL Data TX: Handle 0 flags 0x00 dlen 11                                                                                                                                                                                                                               #156 5407.716000
      ATT: Handle Value Indication (0x1d) len 6
        Handle: 0x9ac9
          Data: 12001c00
> ACL Data RX: Handle 0 flags 0x02 dlen 9                                                                                                                                                                                                                                #157 5407.753700
      ATT: Write Request (0x12) len 4
        Handle: 0x0004
          Data: 0200
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                                                #158 5407.753900
      ATT: Error Response (0x01) len 4
        Write Request (0x12)
        Handle: 0x0000
        Error: Reserved (0x12)
> ACL Data RX: Handle 0 flags 0x02 dlen 5                                                                                                                                                                                                                                #163 5407.808200
      ATT: Handle Value Confirmation (0x1e) len 0
```

```
uart:~$ bt connect dd:ad:fb:8f:b7:ae (random)
Connection pending
Connected: dd:ad:fb:8f:b7:ae (random)
Unsubscribed
```

ioannisg, 2019-08-13:
@MariuszSkamra @Vudentz is this bug in master branch as well?

ioannisg, 2019-08-13:
Bug in a new feature (not regression) in a configuration not enabled by default

Vudentz, 2019-08-14:
We don't support robust caching, or any caching, as GATT client so I wonder what is going one here?

MariuszSkamra, 2019-08-16:
Closing because GATT Caching client side is not supported yet.

closed 2019-08-16 by MariuszSkamra