zephyrproject-rtos/zephyr #18198: SDK 0.10.2 rv32m1_vega samples/subsys/logging/logger build fails

pabigot, 2019-08-12:
This shows up in CI when modifying a file that triggers a test of this sample/platform, but is reproducible on master at a8b208aff77fddd4e02589ed00323a0921a7d77c with:

```
cd ${ZEPHYR_BASE}/samples/subsys/logging/logger
rm -rf r5 && west build -d r5 -b rv32m1_vega_ri5cy
```
The result is:
```
[108/113] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && /usr/local/zephyr-sdk/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc    (blah blah) && :
Memory region         Used Size  Region Size  %age Used
             ROM:       28096 B    1048320 B      2.68%
         VECTORS:         164 B        256 B     64.06%
             RAM:        8784 B       192 KB      4.47%
        IDT_LIST:         249 B         2 KB     12.16app/libapp.a(ext_log_system_adapter.c.obj):
 in function `ext_log_system_log_adapt':
/mnt/nordic/zp/zephyr/samples/subsys/logging/logger/r5/../src/ext_log_system_adapter.c:36:
(.text.ext_log_system_log_adapt+0x0): relocation truncated to fit: R_RISCV_RVC_LUI against `log_handler'
collect2: error: ld returned 1 exit status
```
It appears SDK 0.10.2 has a binutils patch for RISC-V to fix LUI relocation problems.  I don't see any newer patches, but this one may be missing a case.

galak, 2019-08-14:
Being worked on here:

https://github.com/riscv/riscv-binutils-gdb/issues/173

pabigot, 2019-08-17:
Closed with the update to SDK 0.10.3.

closed 2019-08-17 by pabigot