zephyrproject-rtos/zephyr #18202: Disable Duplicate scan, no longer available

Iotfabula, 2019-08-12:
Unable to disable duplicate scan filter, before latest commit ( Whitelist feature ), filter
could be disabled by declaring scan with BT_HCI_LE_SCAN_FILTER_DUP_DISABLE parameter, now
semantic changed, and this option no longer available

File affected:
include/bluetooth/bluetooth.h

**Expected behavior**
User should be enable disable duplicate filter, those enable to see every single BLE advertisements

**Impact**
New Whitelist feature, somehow make this filter always on, preventing our board to see
all adv packets

Add any other context about the problem here.

aescolar, 2019-08-13:
CC @jhedberg

joerchan, 2019-08-14:
The option is still there. If no filtering on scan is needed assign `filter_dup` to `0`
```
struct bt_le_scan_params *scan_param =
			  BT_LE_SCAN_PARAM(BT_HCI_LE_SCAN_ACTIVE, \
					   0, \
					   BT_GAP_SCAN_FAST_INTERVAL, \
					   BT_GAP_SCAN_FAST_WINDOW
```
BTMON:
```
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                               #45 [hci0] 9.180057
        Scanning: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
```

joerchan, 2019-08-15:
Since BT_HCI_LE_SCAN_FILTER_DUP_DISABLE = 0x00 this option can still be used as well.

closed 2019-08-15 by joerchan