zephyrproject-rtos/zephyr #18205: tests/net/socket/udp fails when code coverage is enabled on qemu_x86

camicarballo, 2019-08-12:
This references #17323 

This test fails when code coverage is enabled on qemu_x86 with a "CPU Page Fault" (see below)

```
SeaBIOS (version rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS build v2.0.0-rc1-9-ga8b208aff77f *****
Running test suite socket_udp
===================================================================
starting test - test_send_recv_2_sock
PASS - test_send_recv_2_sock
===================================================================
starting test - test_v4_sendto_recvfrom
PASS - test_v4_sendto_recvfrom
===================================================================
starting test - test_v6_sendto_recvfrom
PASS - test_v6_sendto_recvfrom
===================================================================
starting test - test_v4_bind_sendto
PASS - test_v4_bind_sendto
===================================================================
starting test - test_v6_bind_sendto
PASS - test_v6_bind_sendto
===================================================================
starting test - test_so_priority
PASS - test_so_priority
===================================================================
starting test - test_so_txtime
PASS - test_so_txtime
===================================================================
starting test - test_v4_sendmsg_recvfrom
PASS - test_v4_sendmsg_recvfrom
===================================================================
starting test - test_v6_sendmsg_recvfrom
PASS - test_v6_sendmsg_recvfrom
===================================================================
starting test - test_v4_sendmsg_recvfrom_connected
PASS - test_v4_sendmsg_recvfrom_connected
===================================================================
starting test - test_v6_sendmsg_recvfrom_connected
PASS - test_v6_sendmsg_recvfrom_connected
===================================================================
starting test - setup_eth
PASS - setup_eth
===================================================================
starting test - test_v6_sendmsg_with_txtime
FATAL: ***** CPU Page Fault (error code 0x00000000)
FATAL: Supervisor thread read address 0xff112233
FATAL: PDE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
FATAL: PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
FATAL: eax: 0x00570930, ebx: 0x0000d204, ecx: 0x00000004, edx: 0x00000000
FATAL: esi: 0xff112233, edi: 0x00570930, ebp: 0x00552c24, esp: 0x00552c18
FATAL: eflags: 0x00000202 cs: 0x0008 cr3: 0x0052d3e0
FATAL: call trace:
FATAL: eip: 0x00100390
FATAL:      0x0013b18b (0x570930)
FATAL:      0x0013b61c (0x5725cc)
FATAL:      0x0012a9ba (0x5725cc)
FATAL:      0x0012acdc (0x5725cc)
FATAL:      0x0012ba39 (0x52d480)
FATAL:      0x0012bde7 (0x52d480)
FATAL:      0x00112c35 (0x52d480)
FATAL:      0x00116a9c (0x52d480)
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x0052e020 (unknown)
FATAL: Halting system
```

This can be reproduced by running:
` ./scripts/sanitycheck --coverage -p qemu_x86 -T tests/net/ `

```
$ ./scripts/sanitycheck --coverage -p qemu_x86 -T tests/net/
JOBS: 4
Cleaning output directory /home/camelot/zephyrproject/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
52 tests selected, 21815 tests discarded due to filters
total complete:   43/  52  82%  failed:    0

qemu_x86                  tests/net/socket/udp/net.socket.udp                FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/udp/net.socket.udp/handler.log

total complete:   52/  52  100%  failed:    1
Generating coverage files...
HTML report generated: /home/camelot/zephyrproject/zephyr/sanity-out/coverage/index.html
51 of 52 tests passed with 0 warnings in 570 seconds
```

After running with gdb, it seems that the test fails on the function call in line 721 of tests/net/socket/udp/src/main.c

andrewboie, 2019-08-12:
Interesting that this doesn't fail when --coverage is not used.
Does this fail at all with mps2_an385?

camicarballo, 2019-08-13:
> Interesting that this doesn't fail when --coverage is not used.
> Does this fail at all with mps2_an385?

It does, in fact, fail with mps2_an385 when --coverage is used 

```
***** Booting Zephyr OS build v2.0.0-rc1-9-ga8b208aff77f *****
Running test suite socket_udp
===================================================================
starting test - test_send_recv_2_sock
PASS - test_send_recv_2_sock
===================================================================
starting test - test_v4_sendto_recvfrom
PASS - test_v4_sendto_recvfrom
===================================================================
starting test - test_v6_sendto_recvfrom
PASS - test_v6_sendto_recvfrom
===================================================================
starting test - test_v4_bind_sendto
PASS - test_v4_bind_sendto
===================================================================
starting test - test_v6_bind_sendto
PASS - test_v6_bind_sendto
===================================================================
starting test - test_so_priority
PASS - test_so_priority
===================================================================
starting test - test_so_txtime
PASS - test_so_txtime
===================================================================
starting test - test_v4_sendmsg_recvfrom
PASS - test_v4_sendmsg_recvfrom
===================================================================
starting test - test_v6_sendmsg_recvfrom
PASS - test_v6_sendmsg_recvfrom
===================================================================
starting test - test_v4_sendmsg_recvfrom_connected
PASS - test_v4_sendmsg_recvfrom_connected
===================================================================
starting test - test_v6_sendmsg_recvfrom_connected
PASS - test_v6_sendmsg_recvfrom_connected
===================================================================
starting test - setup_eth
PASS - setup_eth
===================================================================
starting test - test_v6_sendmsg_with_txtime
FATAL: ***** BUS FAULT *****
FATAL:   Precise data bus error
FATAL:   BFAR Address: 0xff112233
FATAL: r0/a1:  0x2002f9b0  r1/a2:  0xff112233  r2/a3:  0xff112237
FATAL: r3/a4:  0x2002f9af r12/ip:  0x00000019 r14/lr:  0x00039331
FATAL:  xpsr:  0x81000000
FATAL: Faulting instruction address (r15/pc): 0x0000098e
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x200255ac (unknown)
FATAL: Halting system
```

jukkar, 2019-08-13:
This also failed with native_posix which made it easy to figure out the culprit when running with valgrind.

PR #18208 (2019-08-14): tests: net: udp: We need to reserve enough space for the data
We are trying to pass 64-bit value to the driver, but we only
allocate space for an integer. This will not work and will cause
invalid memory access.

Fixes #18205
commit 7da78796 (2019-08-14): tests: net: udp: We need to reserve enough space for the data

We are trying to pass 64-bit value to the driver, but we only
allocate space for an integer. This will not work and will cause
invalid memory access.

Fixes #18205
closed 2019-08-14 by jukkar