zephyrproject-rtos/zephyr #18207: tests/bluetooth/hci_prop_evt fails with code coverage enabled in qemu_x86

camicarballo, 2019-08-12:
This references #17324 

This test fails with code coverage enabled in qemu_x86 with a double fault (see below)

```
SeaBIOS (version rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 9
***** Booting Zephyr OS build v2.0.0-rc1-9-ga8b208aff77f *****
Running test suite test_hci_prop_evt
===================================================================
starting test - test_hci_prop_evt_entry
FATAL: Double Fault
FATAL: eax: 0x00000012, ebx: 0x00536040, ecx: 0x00000001, edx: 0x00000000
FATAL: esi: 0x00000000, edi: 0x00105165, ebp: 0x0052b018, esp: 0x0052b000
FATAL: eflags: 0x00000046 cs: 0x0008 cr3: 0x0051d200
FATAL: call trace:
FATAL: eip: 0x001032a7
FATAL:      0x001288df (0x52de9c)
FATAL:      0x0012ba18 (0x52de9c)
FATAL:      0x0012ab09 (0x52de9c)
FATAL:      0x00129961 (0x52de9c)
FATAL:      0x0012af0e (0x52de9c)
FATAL:      0x0012ce83 (0x52de9c)
FATAL:      0x0012cfd3 (0x52de9c)
FATAL:      0x00119a37 (0x52de9c)
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x0051d320 (BT TX)
[00:00:00.000,000] <dbg> bt_hci_core.bt_hci_driver_register: Registered test
ASSERTION FAIL [z_spin_lock_valid(l)] @ ZEPHYR_BASE/include/spinlock.h:78
	Recursive spinlock
FATAL: eax: 0x00000001, ebx: 0x0052b000, ecx: 0x0000000a, edx: 0x00000000
FATAL: esi: 0x00000000, edi: 0x00105165, ebp: 0x0052f57c, esp: 0x0052f578
FATAL: eflags: 0x00000046 cs: 0x0008 cr3: 0x0051d200
FATAL: call trace:
FATAL: eip: 0x00106af5
FATAL:      0x00128dca (0x138734)
FATAL:      0x0012aad9 (0x52f5c8)
FATAL:      0x00129961 (0x51d180)
FATAL:      0x0012af0e (0x51d180)
FATAL:      0x00125cf1 (0x51d180)
FATAL:      0x0010d6f9 (0x51d180)
FATAL:      0x0010d896 (0x5369b8)
FATAL:      0x0010c6d8 (0x5369b8)
FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic
FATAL: Current thread: 0x0051d320 (BT TX)
[00:00:00.010,000] <dbg> bt_hci_core.bt_hci_cmd_create: opcode 0x1003 param_len 0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ZEPHYR_BASE/include/spinlock.h:78
	Recursive spinlock
FATAL: eax: 0x00000002, ebx: 0x0052f578, ecx: 0x0000000a, edx: 0x00000000
FATAL: esi: 0x00000000, edi: 0x00105165, ebp: 0x0052f34c, esp: 0x0052f348
FATAL: eflags: 0x00000046 cs: 0x0008 cr3: 0x0051d200
FATAL: call trace:
FATAL: eip: 0x00106af5
FATAL:      0x00125299 (0x138048)
FATAL:      0x00125cc6 (0x52f39c)
FATAL:      0x0010d6f9 (0x51d180)
FATAL:      0x0010d896 (0x536998)
FATAL:      0x0010c6d8 (0x536998)
FATAL:      0x0010c9a9 (0x536998)
FATAL:      0x0010c46f (0x0)
FATAL:      0x0012401a (0x136a3a)
FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic
FATAL: Current thread: 0x0051d320 (BT TX)
FATAL: Double Fault
FATAL: eax: 0x0010e3cd, ebx: 0x00000001, ecx: 0x000003e8, edx: 0x00000000
FATAL: esi: 0x00000000, edi: 0x00105165, ebp: 0x0052f05c, esp: 0x0052f000
FATAL: eflags: 0x00000016 cs: 0x0008 cr3: 0x0051d200
FATAL: call trace:
FATAL: eip: 0x0010e3cd
FATAL:      0x0010e54c (0x10e3cd)
[... 217 lines cut ...]
[FATAL: Halting system
```

This can be reproduced by running 
`./scripts/sanitycheck --coverage -p qemu_x86 -T tests/bluetooth/`

```
$ ./scripts/sanitycheck --coverage -p qemu_x86 -T tests/bluetooth/
JOBS: 4
Cleaning output directory /home/camelot/zephyrproject/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
16 tests selected, 10425 tests discarded due to filters
total complete:    0/  16   0%  failed:    0

qemu_x86                  tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt/handler.log

total complete:   16/  16  100%  failed:    1
Generating coverage files...
HTML report generated: /home/camelot/zephyrproject/zephyr/sanity-out/coverage/index.html
15 of 16 tests passed with 0 warnings in 91 seconds
```

After running with gdb, it seems to be failing on line 339 of subsys/bluetooth/host/hci_core.c

andrewboie, 2019-08-12:
This looks like it might be some kind of stack overflow.
Can you reproduce with mps2_an385 with --coverage?

camicarballo, 2019-08-13:
> This looks like it might be some kind of stack overflow.
> Can you reproduce with mps2_an385 with --coverage?

This test does not fail with coverage enabled on mps2_an385

carlescufi, 2020-03-16:
@joerchan can you please take a look?

joerchan, 2020-03-17:
Bluetooth TX stack was overflowing. 
Adding these options would fix the problem:
```
CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y
CONFIG_BT_HCI_TX_STACK_SIZE=1500
```

However, it appears enabling QEMU_X86 with code coverage has a severe stack size increase, almost twice the amount of stack use for all stacks. Enabling code coverage would break many optimized stack sizes in the system.
Would code coverage with native_posix be better perhaps? The test passes without modifications then.

QEMU Code Coverage = y
```
I: Stack analyze:
I:  BT RX               : unused 216 usage 808 / 1024 (78 %)
I:  BT TX               : unused 796 usage 1252 / 2048 (61 %)
I:  stack_size_analyzer : unused 124 usage 900 / 1024 (87 %)
I:  sysworkq            : unused 3304 usage 792 / 4096 (19 %)
I:  idle                : unused 2020 usage 28 / 2048 (1 %)
```

QEMU Code Coverage = n
```
I: Stack analyze:
I:  BT RX               : unused 608 usage 416 / 1024 (40 %)
I:  BT TX               : unused 1708 usage 340 / 2048 (16 %)
I:  stack_size_analyzer : unused 652 usage 372 / 1024 (36 %)
I:  sysworkq            : unused 628 usage 396 / 1024 (38 %)
I:  idle                : unused 228 usage 28 / 256 (10 %
```

Also shameless advertising to have a look at https://github.com/zephyrproject-rtos/zephyr/pull/23258 which provided me with these numbers.

joerchan, 2020-03-17:
Closing this issue without fixing the build configuration.

Running these tests with qemu_x86 will require increasing stack sizes for multiple tests.
Running code coverage with native_posix is recommended instead. This test passes for native_posix with code coverage enabled.

closed 2020-03-17 by joerchan