zephyrproject-rtos/zephyr #18231: MCUBoot not cleaning up properly before booting Zephyr?

endian-benjamin, 2019-08-13:
If you enable UART or RTT logging in MCUBoot, it seems these won't be uninitialized after booting to Zephyr, at least on nRF52.

Steps to reproduce the behavior:
1. Build a sample for nRF52 that implements proper power management (suspending all UARTs etc) and flash it "as usual" i.e. at offset `0x00` with bootloader disabled
2. Observe that power consumption is expectedly low when UARTs are disabled, about 4 uA hopefully
3. Build/flash mcuboot with UART enabled and redo step 1, flashing at `slot-0`
4. Observe that power consumption of `slot-0` application behaves as if UART never gets suspended.

Expected behavior:
- `slot-0` application behaves the same wrt power consumption regardless if you're bootloading or not.

Impact:
- If one does not rely upon UART inside bootloader, simply disabling it in the mcuboot `prj.conf` seems like a fine workaround. But I figured it was worth reporting anyway in case this is an indication that far worse bugs could be possible.

ioannisg, 2019-08-13:
Requires some investigation - @nvlsianpu could you help here?

nvlsianpu, 2019-08-14:
@endian-benjamin Thanks for reporting this problem!

I believe that proper place for report such issue is [mcuboot repo](https://github.com/JuulLabs-OSS/mcuboot/issues/new) - this is mcuboot issue.

as far i can tell there is no serial de-initialization before the  application call in mcuboot - so this is the issue. Also zephyr doesn't help MCUBoot much - real device power management is not available jet.

nvlsianpu, 2019-08-14:
In  bootstrapped zephyr application you can try to reinitialize UART which should be kind of a workaround for this problem.

closed 2019-08-14 by nvlsianpu

endian-benjamin, 2019-08-14:
@nvlsianpu Uniting UART before initing sounds robust enough, but I'm not sure I agree this is more suitable for the mcuboot repo - since one of the primary (THE primary?) design goals of mcuboot is to minimize surface area for bugs by being as dumb and static as possible, I'd suspect they'd say this is better done by the bootstrapped application (I'm inclined to agree).

nvlsianpu, 2019-08-14:
Actually I'm one of mcuboot maintainer - I can said that this should be done by mcuboot - we had similar issues for USB for instance and now it is disabled by MCUBoot itself.