zephyrproject-rtos/zephyr #18246: Build failures with current tree

galak, 2019-08-14:
We have the following build failures:

```
warp7_m4                  tests/lib/sprintf/libraries.libc                   FAILED: build_error
        see: sanity-out/warp7_m4/tests/lib/sprintf/libraries.libc/build.log

total complete:    1/  11   9%  failed:    1

96b_meerkat96             tests/lib/sprintf/libraries.libc                   FAILED: build_error
        see: sanity-out/96b_meerkat96/tests/lib/sprintf/libraries.libc/build.log

total complete:    2/  11  18%  failed:    2

colibri_imx7d_m4          tests/lib/sprintf/libraries.libc                   FAILED: build_error
        see: sanity-out/colibri_imx7d_m4/tests/lib/sprintf/libraries.libc/build.log

total complete:    3/  11  27%  failed:    3

rv32m1_vega_ri5cy         samples/subsys/logging/logger/samples.logger       FAILED: build_error
        see: sanity-out/rv32m1_vega_ri5cy/samples/subsys/logging/logger/samples.logger/build.log

total complete:    4/  11  36%  failed:    4

bbc_microbit              samples/bluetooth/mesh_demo/sample.bluetooth.mesh_demo FAILED: build_error
        see: sanity-out/bbc_microbit/samples/bluetooth/mesh_demo/sample.bluetooth.mesh_demo/build.log

total complete:    5/  11  45%  failed:    5

nucleo_l4r5zi             samples/subsys/nvs/sample.subsys.nvs               FAILED: build_error
        see: sanity-out/nucleo_l4r5zi/samples/subsys/nvs/sample.subsys.nvs/build.log

total complete:    6/  11  54%  failed:    6

bbc_microbit              samples/bluetooth/hci_uart/sample.bluetooth.hci_uart.arm FAILED: build_error
        see: sanity-out/bbc_microbit/samples/bluetooth/hci_uart/sample.bluetooth.hci_uart.arm/build.log

total complete:    9/  11  81%  failed:    7

nrf52810_pca10040         tests/kernel/mutex/sys_mutex/sys.mutex             FAILED: build_error
        see: sanity-out/nrf52810_pca10040/tests/kernel/mutex/sys_mutex/sys.mutex/build.log

total complete:   10/  11  90%  failed:    8

nrf52811_pca10056         tests/kernel/mutex/sys_mutex/sys.mutex             FAILED: build_error
        see: sanity-out/nrf52811_pca10056/tests/kernel/mutex/sys_mutex/sys.mutex/build.log
```

galak, 2019-08-14:
The majority of these are code or memory size issues.

The rv32m1_vega_ri5cy is reported in #18198

the nucleo_l4r5zi is a compile error:

```
/home/galak/git/FULL_ZEPHYR/zephyr/drivers/flash/flash_stm32l4x.c: In function 'flash_stm32_block_erase_loop':
/home/galak/git/FULL_ZEPHYR/zephyr/drivers/flash/flash_stm32l4x.c:149:11: error: 'pages_per_bank' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  if (page >= pages_per_bank)
      ~~~~~^~~~~~~~~~~~~~~~~
/home/galak/git/FULL_ZEPHYR/zephyr/drivers/flash/flash_stm32l4x.c:118:8: note: 'pages_per_bank' was declared here
  u16_t pages_per_bank;
        ^~~~~~~~~~~~~~
```

carlescufi, 2019-08-15:
@ioannisg I took a look at the last two failures with Nordic boards and `tests/kernel/mutex/sys_mutex/sys.mutex`. I am not sure whether the right fix is to restrict that test to platforms with at least 32KB of RAM or reduce the ISR stack size from 2048 to 1024 or similar.

carlescufi, 2019-08-15:
> @ioannisg I took a look at the last two failures with Nordic boards and `tests/kernel/mutex/sys_mutex/sys.mutex`. I am not sure whether the right fix is to restrict that test to platforms with at least 32KB of RAM or reduce the ISR stack size from 2048 to 1024 or similar.

I have created #18295 to fix this by requiring 32KB of RAM which made sense to me.

ioannisg, 2019-08-16:
> I have created #18295 to fix this by requiring 32KB of RAM which made sense to me.

Merged now @carlescufi

ioannisg, 2019-08-16:
@galak the bbc_microbit fail on hci_uart samples does not appear (anymore), at least in my local sanitycheck run

carlescufi, 2019-08-16:
> @galak the bbc_microbit fail on hci_uart samples does not appear (anymore), at least in my local sanitycheck run
> 
> Edit: same for the bluetooth.mesh_demo - I can't make it fail.

I fixed those in https://github.com/zephyrproject-rtos/zephyr/pull/18294

PR #18337 (2019-08-17): drivers/flash: stm32l4: Handle pages_per_bank for all variants
Variable pages_per_bank has been introduced to ease page erase
in dual bank configurations. This has been implemented using
FLASH_OPTR_DUALBANK definition.
It happen that this was not taking into account L4+ series that
use FLASH_OPTR_DBANK instead of FLASH_OPTR_DUALBANK. This lead
to compilation issue for this driver in case of L4+ series.
So, this patch is adding the support of FLASH_OPTR_DBANK definition.

Besides, FLASH_OPTR_DUALBANK (as FLASH_OPTR_DBANK) are actually
defining availability of an option byte to configure use of  [... cut]
commit 0381e8cb (2019-08-17): drivers/flash: stm32l4: Handle pages_per_bank for all variants

Variable pages_per_bank has been introduced to ease page erase
in dual bank configurations. This has been implemented using
FLASH_OPTR_DUALBANK definition.
It happen that this was not taking into account L4+ series that
use FLASH_OPTR_DBANK instead of FLASH_OPTR_DUALBANK. This lead
to compilation issue for this driver in case of L4+ series.
So, this patch is adding the support of FLASH_OPTR_DBANK definition.

Besides, FLASH_OPTR_DUALBANK (as FLASH_OPTR_DBANK) are actually
defining availability of an option byte to configure use of [... cut]
closed 2019-08-17 by ioannisg