zephyrproject-rtos/zephyr #18261: CONFIG_TIMESLICING=n breaks kernel

pizi-nordic, 2019-08-14:
Disabling time-slicing (which is enabled default) results in crashing threads.

**How to Reproduce:**
Example 1:
  1. $ cmake -GNinja -DBOARD=nrf52840_pca10056 ~/repos/zephyr/samples/hello_world/
  2. $ ninja flash
       ```
       ***** Booting Zephyr OS build v2.0.0-rc1-17-gc2d702b96177 *****
       Hello World! nrf52840_pca10056
       ```
  3. $ sed -i 's/CONFIG_TIMESLICING=y/# CONFIG_TIMESLICING is not set/' zephyr/.config
  4. $ ninja flash
      ```
      ***** Booting Zephyr OS build v2.0.0-rc1-17-gc2d702b96177 *****
      Hello World! nrf52840_pca10056
      FATAL: ***** USAGE FAULT *****
      FATAL:   Illegal load of EXC_RETURN into PC
      FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
      FATAL: Current thread: 0x2000014c (unknown)
      FATAL: Halting system
     ```
Example 2:
  1. $ cmake -GNinja -DBOARD=nrf52840_pca10056 ~/repos/zephyr/tests/kernel/sleep/
  2. $ ninja flash
       ```
       ***** Booting Zephyr OS build v2.0.0-rc1-17-gc2d702b96177 *****
       (...)
       PROJECT EXECUTION SUCCESSFUL
       ```
  3. $ sed -i 's/CONFIG_TIMESLICING=y/# CONFIG_TIMESLICING is not set/' zephyr/.config
  4. $ ninja flash
      ```
      ***** Booting Zephyr OS build v2.0.0-rc1-17-gc2d702b96177 *****
      Running test suite sleep
      ===================================================================
      starting test - test_sleep
      Kernel objects initialized
      Test thread started: id = 0x2000014c
      Helper thread started: id = 0x20000088
      Testing normal expiration of k_sleep()
      ASSERTION FAIL [(u32_t)thread->stack_obj == guard_start] @ ZEPHYR_BASE/arch/arm/core/cortex_m/mpu/arm_core_mpu.c:236
              Guard start (0xffffffe0) not beginning at stack object (0x0)

      FATAL: r0/a1:  0x00000004  r1/a2:  0x000000ec  r2/a3:  0x00000001
      FATAL: r3/a4:  0x0000000e r12/ip:  0x00000000 r14/lr:  0x000021f5
      FATAL:  xpsr:  0x6100000e
      FATAL: Faulting instruction address (r15/pc): 0x00008e60
      FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic
      FATAL: Current thread: 0x20000548 (�∟)
      FATAL: Halting system
      ```

andyross, 2019-08-14:
Is this on nrf only?  I can't make this fail via the sanitycheck platforms.

carlescufi, 2019-08-15:
This is not an issue actually @pizi-nordic. You can't just edit the `.config` file, you need to let Kconfig figure out the dependencies.

I edit the `prj.conf` for both `samples/hello_world` and `tests/kernel/sleep` and added a `CONFIG_TIMESLICING=n` and everything works fine.

pizi-nordic, 2019-08-16:
@carlescufi: If you edit this settings through menuconfig (which should resolve dependencies or do not allow to edit thing which is needed), the problem still exists.

@andyross: I tested this only on nordic.

pizi-nordic, 2019-08-16:
Grrrr. It looks that it fails only on my board. After switching to another problem disappeared...

closed 2019-08-16 by pizi-nordic