zephyrproject-rtos/zephyr #18269: Documentation improvement for macOS

chenguokai, 2019-08-14:
**Describe the bug**

At least for MacPorts users, with a clean environment, the Host Dependencies part is incomplete. Without Rust and cargo, Install Python Dependencies step will fail for lack of cargo.

**To Reproduce**
Steps to reproduce the behavior:

1. Follow the Getting started guide until the pip dependencies part
2. get the error

**Expected behavior**
It should build without any error

**Impact**
Prevent the project from building on macOS

**Screenshots or console output**

The error log:

Building wheels for collected packages: cmsis-pack-manager
  Building wheel for cmsis-pack-manager (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/local/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/tmp/pip-wheel-z866nvuj --python-tag cp37
       cwd: /private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/
  Complete output (9 lines):
  fatal: Not a git repository (or any of the parent directories): .git
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  creating build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
  error: [Errno 2] No such file or directory: 'cargo': 'cargo'
  ----------------------------------------
  ERROR: Failed building wheel for cmsis-pack-manager
  Running setup.py clean for cmsis-pack-manager
Failed to build cmsis-pack-manager
ERROR: pyocd 0.21.0 has requirement cmsis-pack-manager>=0.2.7, but you'll have cmsis-pack-manager 0.1.1 which is incompatible.
Installing collected packages: pyusb, intelhex, cmsis-pack-manager, sortedcontainers, intervaltree, pyocd, pyserial, Pillow, zipp, importlib-metadata, pluggy, py, more-itertools, wcwidth, atomicwrites, pytest, gcovr
  Running setup.py install for cmsis-pack-manager ... error
    ERROR: Command errored out with exit status 1:
     command: /opt/local/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/tmp/pip-record-y3p0vclu/install-record.txt --single-version-externally-managed --compile
         cwd: /private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/
    Complete output (9 lines):
    fatal: Not a git repository (or any of the parent directories): .git
    running install
    running build
    running build_py
    creating build/lib
    creating build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
    error: [Errno 2] No such file or directory: 'cargo': 'cargo'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /opt/local/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/private/tmp/pip-install-4r20yn1r/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/tmp/pip-record-y3p0vclu/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

**Environment (please complete the following information):**
 - OS: macOS 10.12.6 
 - Toolchain: Not used yet

**Additional context**
Similar problem with FreeBSD:
https://github.com/ARMmbed/cmsis-pack-manager/issues/97

aescolar, 2019-08-14:
@chenguokai  . Thanks for the bug report. Would you like to submit a pull request with a proposal of the fixes to the documentation?

carlescufi, 2019-08-14:
@chenguokai I have never installed `cmsis-package-manager` or Rust or cargo on my macOS machine and build Zephyr every day. Macports is not supported officially, only Homebrew, so this is not a bug but rather an improvement

carlescufi, 2019-08-14:
Just double-checked that, with Homebrew's Python no Rust is required at all:

```
Collecting cmsis-pack-manager>=0.2.7 (from pyocd==0.21.0->-r scripts/requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/88/e9/4cd8d5069f805eeac595eaca58ea93cf859532ed24614b6e6d0178a074c7/cmsis_pack_manager-0.2.8-py2.py3-none-macosx_10_13_x86_64.whl (2.1MB)
     |████████████████████████████████| 2.2MB 16.2MB/s 
```

chenguokai, 2019-08-15:
Maybe there are some differences between the python3 installed with MacPorts and Homebrew. I will try to make a pull request adding a comment for this.

carlescufi, 2019-08-15:
> Maybe there are some differences between the python3 installed with MacPorts and Homebrew. I will try to make a pull request adding a comment for this.

Thanks for that @chenguokai. Look forward to reviewing your Pull Request.

PR #18299 (2019-08-16): doc: update macOS specific guide for MacPorts
Modified installation_mac.rst to add a comment for the possible setup failure.
The solution can be infered from ARMmbed/cmsis-pack-manager#97
Fixes #18269.
commit 82da4c57 (2019-08-16): doc: update macOS specific guide for MacPorts

Add comment for the setup failure.
The solution can be infered from ARMmbed/cmsis-pack-manager#97
Fixes #18269.
closed 2019-08-16 by carlescufi