zephyrproject-rtos/zephyr #18273: RFC: list required / optional arch features and clarify policy

andrewboie, 2019-08-14:
We need to identify optional arch features and decide which ones are truly required for acceptance in the tree.

First thing, let's try to put together a list of all these configs or kernel features which may (at least in some cases) require support in arch/ code. This is what I came up with so far, please amend as necessary:

- CONFIG_HAS_DTS
- CONFIG_HW_STACK_PROTECTION
- CONFIG_USERSPACE
- CONFIG_DYNAMIC_INTERRUPTS
- CONFIG_STACK_SENTINEL
- CONFIG_FLOAT / CONFIG_FP_SHARING
- CONFIG_IRQ_OFFLOAD
- CONFIG_INIT_STACKS
- CONFIG_STACK_POINTER_RANDOM
- CONFIG_THREAD_MONITOR
- CONFIG_EXECUTION_BENCHMARKING (can we merge with CONFIG_TRACING?)
- CONFIG_THREAD_USERSPACE_LOCAL_DATA (should be renamed to thread-local storage)
- CONFIG_TRACING
- CONFIG_SYS_POWER_MANAGEMENT (and related configs)
- CONFIG_EXCEPTION_STACK_TRACE (feasibility depends on arch, ARM is very hard for example)
- CONFIG_BOOT_TIME_MEASUREMENT (can we merge with CONFIG_TRACING?)
- CONFIG_COVERAGE_GCOV (XIP considerations for extra gcov sections)
- CONFIG_STACK_CANARIES (may or may not require arch support)
- CONFIG_SMP
- Z_ARCH_EXCEPT() (no associated config, see include/kernel.h)

Once we have this full list, determine what is the *minimum* set (assuming HW support) that has to be implemented before an arch port can be merged into the tree. And then review tests to ensure that these are all being tested.

aescolar, 2019-08-15:
> truly required for acceptance in the tree.

One side comment here: I think it would be more constructive to think it as: What features should have more focus on their documentation, and having a clear API, and appropriate tests, to ensure whoever wants to contribute them, can implement it with a reasonable effort.

andrewboie, 2019-08-15:
> What features should have more focus on their documentation, and having a clear API, and appropriate tests, to ensure whoever wants to contribute them, can implement it with a reasonable effort.

No, that's really not what I'm getting at here at all.

I open this because I keep getting surprised on what features are missing from our arch ports. I want to set a minimum standard of features (it will not be all of what's named above) that are minimum criteria for the arch port to go in. Period.

If you think some features have insufficient documentation, or have badly designed APIs, or have insufficient tests, go file bugs on them specifically.

ioannisg, 2019-10-03:
>CONFIG_INIT_STACKS and stack_analyze() (doesn't work on ARC/ARMv8 when examining active thread)

@andrewboie why is this not working for ARMv8-M? Is there any issue reported? :)

ioannisg, 2019-10-04:
@andrewboie  perhaps the following too?
ATOMIC OPERATIONS
USE_SWITCH
XIP

ioannisg, 2020-01-08:
DYNAMIC_DIRECT_INTERRUPTS, after #21516 is merged