zephyrproject-rtos/zephyr #18276: irq_connect_dynamic() is not tested on all arches

andrewboie, 2019-08-14:
We have very little coverage of installing dynamic interrupts in test cases.

Even though gen_isr_tables.py is used on multiple arches, the test case for it is running only on ARM for some reason:

```
tests:
  arch.interrupt:
    platform_exclude: nucleo_f103rb olimexino_stm32 stm32_min_dev_black
        stm32_min_dev_blue usb_kw24d512 v2m_beetle
    filter: CONFIG_GEN_ISR_TABLES and CONFIG_ARMV7_M_ARMV8_M_MAINLINE
    tags: interrupt isr_table
```
Because of this, bugs like #18272 crept in.

We need to test dynamic IRQs on all arches that support it to show that they work.

aescolar, 2019-08-15:
@andrewboie @rljordan : Given how our release process is, issues related to something not yet being implemented/not yet tested cannot be classified as a bug. Otherwise we would just block all our future releases.
This is independent of how important or relevant the issue is, or how much we would like it to be fixed.
So, I relabel it as an enhancement.

andrewboie, 2019-08-15:
This is *not* an enhancement!

>     filter: CONFIG_GEN_ISR_TABLES and CONFIG_ARMV7_M_ARMV8_M_MAINLINE

The bug is that CONFIG_ARMV7_M_ARMV8_M_MAINLINE should not have been put in the filter for this test, resulting in irq_connect_dynamic() not being tested on arches that support it.

aescolar, 2019-08-15:
@andrewboie Uhm, my bad. I missunderstood the bug description.

ioannisg, 2019-08-16:
@andrewboie the test you mentioned was always ARM-specific, AFAICS :)
Could you please, mention which other ARCHes are supposed to support dynamic interrupts ?

andrewboie, 2019-08-16:
@ioannisg anything that is using gen_isr_tables.py to create their interrupt tables.
Should be all arches aside from x86, I believe. ARM, ARC, RiscV, Nios II, Xtensa all use it.
BTW although this is a bug I don't think it needs to be 'medium' I can drop to low.

closed 2021-01-04 by github-actions[bot]