zephyrproject-rtos/zephyr #18280: tests/drivers/adc/adc_api fails on frdmkl25z

hakehuang, 2019-08-15:
**Describe the bug**
test_adc_sample_with_interval fails

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_kl25z ..
3. make
4. See error

**Expected behavior**
ALL cases PASS

**Impact**
interval adc sample error

**Screenshots or console output**
````
***** Booting Zephyr OS version 2.0.0-rc1 *****

Running test suite adc_basic_test

===================================================================

starting test - test_adc_sample_one_channel

Samples read: 0x0501 0x0000 0x0000 0x0000 0x0000 0x0000 

PASS - test_adc_sample_one_channel

===================================================================

starting test - test_adc_sample_two_channels

SKIP - test_adc_sample_two_channels

===================================================================

starting test - test_adc_asynchronous_call

Samples read: 0x0438 0x0462 0x05b7 0x04c2 0x04dd 0x0000 

PASS - test_adc_asynchronous_call

===================================================================

starting test - test_adc_sample_with_interval

sample_with_interval_callback: sampling 0

sample_with_interval_callback: sampling 1

sample_with_interval_callback: sampling 2

sample_with_interval_callback: sampling 3

sample_with_interval_callback: sampling 4

    Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:389: test_task_with_interval: (ret not equal to 0)

adc_read() failed with code -16

FAIL - test_adc_sample_with_interval

===================================================================

starting test - test_adc_repeated_samplings

repeated_samplings_callback: done 1

Samples read: 0x02a5 0x0000 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 2

Samples read: 0x02a5 0x028d 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 3

Samples read: 0x02a5 0x0211 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 4

Samples read: 0x02a5 0x01a8 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 5

Samples read: 0x02a5 0x013b 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 6

Samples read: 0x02a5 0x0263 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 7

Samples read: 0x02a5 0x0408 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 8

Samples read: 0x02a5 0x03db 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 9

Samples read: 0x02a5 0x037b 0x0000 0x0000 0x0000 0x0000 

repeated_samplings_callback: done 10

Samples read: 0x02a5 0x020b 0x0000 0x0000 0x0000 0x0000 

PASS - test_adc_repeated_samplings

===================================================================

starting test - test_adc_invalid_request

[00:00:00.596,000] [1;31m<err> adc_mcux_adc16: Invalid resolution[0m

[00:00:00.602,000] [1;31m<err> adc_mcux_adc16: Invalid resolution[0m

Samples read: 0x026a 0x0000 0x0000 0x0000 0x0000 0x0000 

PASS - test_adc_invalid_request

===================================================================

Test suite adc_basic_test failed.

===================================================================

PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK, 0.10.2
 - R2.0.0-rc1

hakehuang, 2019-08-15:
@MaureenHelm

dleach02, 2019-08-20:
@MaureenHelm, @hakehuang, I've run this test 20+ times and I've only received the error once. Looking closer at the code it looks like the busy error will occur if the hardware hasn't finished the previous request. Not sure what to do about this but I suggest lowering the priority of this to low unless Hake has some way to get this to reproduce reliably.

hakehuang, 2019-08-20:
@dleach02 , I do not have any good way to reproduce it. and looks like I run multiply cases in batch, then this will happen. maybe we need consider to add a sync read lock in driver

dleach02, 2019-08-20:
@hakehuang, I haven't looked deeply at the API and underlying code but I think there is already a lock in the API level. I'm thinking there is some other transient problem but given how infrequently it occurs it may be difficult to instrument to debug. This is why I'm suggesting we move this to low priority to not block the release.

dleach02, 2019-08-21:
Tried the same test on the frdmk64f and could not get it to fail. Same IP apparently.

hakehuang, 2019-08-29:
@dleach02 do you try to use the on board debuger to download image?

dleach02, 2019-09-03:
@hakehuang, yes, that is how I programmed the board.

hakehuang, 2019-09-04:
@dleach02 may Ihave you board revision? mine is rev D

dleach02, 2019-09-04:
The tag on the back has:

700-27556 REV C
SCH-27556 REV F

I don't see any other labels.

hakehuang, 2019-09-11:
I do a experiment and soem code change below
{code}
	ret = adc_read(adc_dev, &sequence);
	TC_PRINT("ret = %d \n", ret);
	zassert_equal(ret, 0, "adc_read() failed with code %d", ret);
{code}

then the application can pass. looks like the code alignment impacts the performance in kl25 platform.

and something makes me strange is that when I try to diff the assembly code of with TC_PRINRT and without. by /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -D zephyr.elf, and the differences i shuge, which seems not reasonable. a single line of change will change the code structure so much. @dleach02 do you know why?

closed 2020-02-14 by hakehuang