zephyrproject-rtos/zephyr #18281: tests/kernel/mem_protect/protection fails on LPC54114_m4

hakehuang, 2019-08-15:
**Describe the bug**
exec_data, exec_heap, exec_stack does not trigger protection

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso54114_m4 ..
3. make
4. See error

**Expected behavior**
All PASS

**Impact**
protection features are not fully supported

**Screenshots or console output**
```
***** Booting Zephyr OS version 2.0.0-rc1 *****

Running test suite protection

===================================================================

starting test - exec_data

trying to call code written to 0x20000009

returned from code at 0x20000009

Execute from target buffer succeeded!

    Assertion failed at ZEPHYR_BASE/tests/kernel/mem_protect/protection/src/main.c:140: exec_data: (Reached unreachable code)

Execute from data did not fault

FAIL - exec_data

===================================================================

starting test - exec_stack

trying to call code written to 0x20000729

returned from code at 0x20000729

Execute from target buffer succeeded!

    Assertion failed at ZEPHYR_BASE/tests/kernel/mem_protect/protection/src/main.c:153: exec_stack: (Reached unreachable code)

Execute from stack did not fault

FAIL - exec_stack

===================================================================

starting test - exec_heap

trying to call code written to 0x200001fd

returned from code at 0x200001fd

Execute from target buffer succeeded!

    Assertion failed at ZEPHYR_BASE/tests/kernel/mem_protect/protection/src/main.c:168: exec_heap: (Reached unreachable code)

Execute from heap did not fault

FAIL - exec_heap

===================================================================

starting test - write_ro

trying to write to rodata at 0x00004614

FATAL: ***** BUS FAULT *****

FATAL:   Imprecise data bus error

FATAL: r0/a1:  0x00000000  r1/a2:  0x0000000a  r2/a3:  0x80000000

FATAL: r3/a4:  0x0ff00ff0 r12/ip:  0x00000000 r14/lr:  0x0000093d

FATAL:  xpsr:  0x61000000

FATAL: Faulting instruction address (r15/pc): 0x00000948

FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception

FATAL: Current thread: 0x2000004c (unknown)

Caught system error -- reason 0

PASS - write_ro

===================================================================

starting test - write_text

trying to write to text at 0x00000a24

FATAL: ***** BUS FAULT *****

FATAL:   Imprecise data bus error

FATAL: r0/a1:  0x00000a24  r1/a2:  0x000038f3  r2/a3:  0x00003902

FATAL: r3/a4:  0x00000a24 r12/ip:  0x00000000 r14/lr:  0x000009a7

FATAL:  xpsr:  0x81000200

FATAL: Faulting instruction address (r15/pc): 0x0000400e

FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception

FATAL: Current thread: 0x2000004c (unknown)

Caught system error -- reason 0

PASS - write_text

===================================================================

Test suite protection failed.

===================================================================

PROJECT EXECUTION FAILED

```
**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain  Zephyr SDK
 - R2.0.0-rc1

hakehuang, 2019-08-15:
@MaureenHelm

ioannisg, 2019-08-18:
@hakehuang this test has the following filter:
`filter: CONFIG_MEMORY_PROTECTION
`
Now, MEMORY_PROTECTION requires ARM_MPU, which is not defined for this platform. So the test should not run for this board at all.

@MaureenHelm I am closing this one.

closed 2019-08-18 by ioannisg