zephyrproject-rtos/zephyr #18282: tests/kernel/sched/schedule_api/ fails on LPC54114_m4

hakehuang, 2019-08-15:
**Describe the bug**
test_slice_scheduling fails

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=LPC54114_M4 ..
3. make
4. See error

**Expected behavior**
PASS

**Impact**
slice_scheduling

**Screenshots or console output**
```
***** Booting Zephyr OS version 2.0.0-rc1 *****

Running test suite threads_scheduling

===================================================================

starting test - test_bad_priorities

PASS - test_bad_priorities

===================================================================

starting test - test_priority_cooperative

PASS - test_priority_cooperative

===================================================================

starting test - test_priority_preemptible

PASS - test_priority_preemptible

===================================================================

starting test - test_yield_cooperative

PASS - test_yield_cooperative

===================================================================

starting test - test_sleep_cooperative

PASS - test_sleep_cooperative

===================================================================

starting test - test_sleep_wakeup_preemptible

PASS - test_sleep_wakeup_preemptible

===================================================================

starting test - test_pending_thread_wakeup

PASS - test_pending_thread_wakeup

===================================================================

starting test - test_time_slicing_preemptible

PASS - test_time_slicing_preemptible

===================================================================

starting test - test_time_slicing_disable_preemptible

PASS - test_time_slicing_disable_preemptible

===================================================================

starting test - test_lock_preemptible

PASS - test_lock_preemptible

===================================================================

starting test - test_unlock_preemptible

PASS - test_unlock_preemptible

===================================================================

starting test - test_unlock_nested_sched_lock

PASS - test_unlock_nested_sched_lock

===================================================================

starting test - test_sched_is_preempt_thread

PASS - test_sched_is_preempt_thread

===================================================================

starting test - test_slice_reset

PASS - test_slice_reset

===================================================================

starting test - test_slice_scheduling

ABCDEFGHI

ABCDEFGHI

ABCDEFGHI

A

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

B

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

C

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

D

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

E

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

F

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

G

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

H

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

I

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:63: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

FAIL - test_slice_scheduling

===================================================================

starting test - test_priority_scheduling

PASS - test_priority_scheduling

===================================================================

starting test - test_wakeup_expired_timer_thread

PASS - test_wakeup_expired_timer_thread

===================================================================

starting test - test_user_k_wakeup

PASS - test_user_k_wakeup

===================================================================

starting test - test_user_k_is_preempt

PASS - test_user_k_is_preempt

===================================================================

Test suite threads_scheduling failed.

===================================================================

PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - R2.0.0-rc1

hakehuang, 2019-08-15:
@MaureenHelm

ioannisg, 2019-08-19:
I think this should be a random (sporadic) fail of the notorious time-slicing test, and nothing to do with this board. It passes for me, locally, on the same board.

Can we have @andyross look at the scheduling test?

agansari, 2019-08-19:
Can't reproduce this as well, it maybe spuroius as @ioannisg mentions.
@hakehuang is there any other detail in reproducing this issue?

ioannisg, 2019-08-19:
@hakehuang could youhelp us a bit more in this issue?

hakehuang, 2019-08-20:
@ioannisg , I just download it to board and run it.  let me try to find a easy way to reproduce it.

agansari, 2019-08-22:
I've reproduced this issue by doing `git checkout v2.0.0-rc1`
It does pass on the latest codebase.

ioannisg, 2019-08-22:
Same results for me. I am closing this, then. Please, re-open if needed, @hakehuang .

ioannisg, 2019-10-09:
@agansari this is, once more, not reproducible, on current master. What's your take - shall we close the issue again? Could you try it yourself as well?

agansari, 2019-10-09:
@ioannisg as i last remember it's related to #18651, this means it will not reproduce every time.
Will try to reproduce this issue again today.

agansari, 2019-10-09:
Tested 5 times on master branch, will be closed as it passes.

closed 2019-10-09 by agansari