zephyrproject-rtos/zephyr #18283: tests/crypto/tinycrypt_hmac_prng  build failure on mimxrt1015_evk

hakehuang, 2019-08-15:
**Describe the bug**
build reports size issue

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1015_evk ..
3. make
4. See error

**Expected behavior**
PASS

**Impact**
tinycrypt

**Screenshots or console output**
```
-- Build files have been written to: /build/src/workspace/rt1015_master/zephyr/tests/crypto/tinycrypt_hmac_prng/build
Scanning dependencies of target kobj_types_h_target
[  1%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[  1%] Built target kobj_types_h_target
Scanning dependencies of target syscall_macros_h_target
[  2%] Generating include/generated/syscall_macros.h
[  2%] Built target syscall_macros_h_target
Scanning dependencies of target syscall_list_h_target
[  3%] Generating misc/generated/syscalls.json
[  4%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  5%] Built target syscall_list_h_target
Scanning dependencies of target driver_validation_h_target
[  6%] Generating include/generated/driver-validation.h
[  6%] Built target driver_validation_h_target
Scanning dependencies of target offsets
[  7%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  7%] Built target offsets
Scanning dependencies of target offsets_h
[  8%] Generating include/generated/offsets.h
[  8%] Built target offsets_h
Scanning dependencies of target app
[  8%] Building C object CMakeFiles/app.dir/src/hmac_prng.c.obj
[  9%] Building C object CMakeFiles/app.dir/src/main.c.obj
[ 10%] Linking C static library app/libapp.a
[ 10%] Built target app
Scanning dependencies of target kernel
[ 11%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj
[ 12%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj
[ 13%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
[ 14%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj
[ 14%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj
[ 16%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj
[ 18%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
[ 19%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[ 20%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[ 21%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[ 21%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj
[ 22%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[ 23%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
[ 24%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
[ 24%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj
[ 25%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
[ 26%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
[ 27%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
[ 28%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[ 28%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[ 29%] Linking C static library libkernel.a
[ 29%] Built target kernel
Scanning dependencies of target linker_script_target
[... 100 lines cut ...]
Scanning dependencies of target zephyr_prebuilt
[ 94%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 95%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       86532 B        16 MB      0.52%
            SRAM:       44776 B        32 KB    136.65%
        IDT_LIST:          72 B         2 KB      3.52/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000020008000,0000000020008047] overlaps section noinit VMA [0000000020000260,000000002000ae1f]
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 12008 bytes
collect2: error: ld returned 1 exit status
%
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:97: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:544: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
script returned exit code 2

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - R2.0.0-rc1

hakehuang, 2019-08-15:
@MaureenHelm

hakehuang, 2019-08-15:
build log for tests/posix/fs

```
-- Configuring done
-- Generating done
-- Build files have been written to: /build/src/workspace/rt1015_kernel_master/zephyr/tests/posix/fs/build
Scanning dependencies of target kobj_types_h_target
[  1%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[  1%] Built target kobj_types_h_target
Scanning dependencies of target syscall_macros_h_target
[  2%] Generating include/generated/syscall_macros.h
[  2%] Built target syscall_macros_h_target
Scanning dependencies of target syscall_list_h_target
[  3%] Generating misc/generated/syscalls.json
[  3%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  4%] Built target syscall_list_h_target
Scanning dependencies of target driver_validation_h_target
[  5%] Generating include/generated/driver-validation.h
[  5%] Built target driver_validation_h_target
Scanning dependencies of target offsets
[  6%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  6%] Built target offsets
Scanning dependencies of target offsets_h
[  6%] Generating include/generated/offsets.h
[  6%] Built target offsets_h
Scanning dependencies of target app
[  6%] Building C object CMakeFiles/app.dir/src/main.c.obj
[  7%] Building C object CMakeFiles/app.dir/src/test_fat_mount.c.obj
[  8%] Building C object CMakeFiles/app.dir/src/test_fs_dir.c.obj
[  8%] Building C object CMakeFiles/app.dir/src/test_fs_file.c.obj
[  9%] Linking C static library app/libapp.a
[  9%] Built target app
Scanning dependencies of target kernel
[ 10%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj
[ 10%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj
[ 11%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
[ 12%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj
[ 12%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[ 13%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj
[ 14%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj
[ 16%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[ 18%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[ 19%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj
[ 19%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[ 20%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
[ 21%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
[ 21%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj
[ 22%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
[ 23%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
[ 23%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
[ 24%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[ 25%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[ 25%] Linking C static library libkernel.a
[ 25%] Built target kernel
Scanning dependencies of target linker_script_target
[ 26%] Generating linker.cmd
[ 26%] Built target linker_script_target
Scanning dependencies of target zephyr
[ 27%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/common/isr_tables.c.obj
[ 27%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/common/sw_isr_common.c.obj
[ 28%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[ 29%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[ 29%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[ 30%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[ 31%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[ 31%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[... 106 lines cut ...]
[ 92%] Built target drivers__serial
Scanning dependencies of target ..__modules__fs__fatfs
[ 92%] Building C object modules/fatfs/CMakeFiles/..__modules__fs__fatfs.dir/ff.c.obj
[ 93%] Building C object modules/fatfs/CMakeFiles/..__modules__fs__fatfs.dir/zfs_diskio.c.obj
[ 94%] Linking C static library lib..__modules__fs__fatfs.a
[ 94%] Built target ..__modules__fs__fatfs
Scanning dependencies of target zephyr_prebuilt
[ 95%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 96%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       59500 B        16 MB      0.35%
            SRAM:       95036 B        32 KB    290.03%
        IDT_LIST:          72 B         2 KB      3.52/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr_prebuilt.elf section `bss' will not fit in region `SRAM'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000020008000,0000000020008047] overlaps section bss VMA [0000000020000000,00000000200147c1]
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 62268 bytes
collect2: error: ld returned 1 exit status
%
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:100: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:550: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [all] Error 2
Makefile:83: recipe for target 'all' failed
script returned exit code 2
```

hakehuang, 2019-08-15:
this is the log for tests/crypto/mbedtls

```
-- Configuring done
-- Generating done
-- Build files have been written to: /build/src/workspace/rt1015_master/zephyr/tests/crypto/mbedtls/build
Scanning dependencies of target syscall_macros_h_target
[  0%] Generating include/generated/syscall_macros.h
[  0%] Built target syscall_macros_h_target
Scanning dependencies of target kobj_types_h_target
[  1%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[  1%] Built target kobj_types_h_target
Scanning dependencies of target driver_validation_h_target
[  1%] Generating include/generated/driver-validation.h
[  1%] Built target driver_validation_h_target
Scanning dependencies of target syscall_list_h_target
[  2%] Generating misc/generated/syscalls.json
[  2%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  3%] Built target syscall_list_h_target
Scanning dependencies of target offsets
[  4%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  4%] Built target offsets
Scanning dependencies of target offsets_h
[  4%] Generating include/generated/offsets.h
[  4%] Built target offsets_h
Scanning dependencies of target app
[  5%] Building C object CMakeFiles/app.dir/src/main.c.obj
[  5%] Building C object CMakeFiles/app.dir/src/mbedtls.c.obj
[  6%] Linking C static library app/libapp.a
[  6%] Built target app
Scanning dependencies of target kernel
[  7%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj
[  7%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj
[  7%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
[  8%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj
[  8%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[  9%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj
[  9%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj
[  9%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[ 10%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj
[ 10%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
[ 11%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[ 11%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[ 11%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[ 12%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj
[ 12%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[ 13%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
[ 13%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
[ 13%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj
[ 14%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
[ 14%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[ 15%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[ 16%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/futex.c.obj
[ 16%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_domain.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/userspace_handler.c.obj
[ 17%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/userspace.c.obj
[ 17%] Linking C static library libkernel.a
[ 17%] Built target kernel
Scanning dependencies of target zephyr
[ 17%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/common/isr_tables.c.obj
[ 18%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/common/sw_isr_common.c.obj
[ 18%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[ 19%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[ 19%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[ 19%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[ 20%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[ 20%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[ 21%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/mempool.c.obj
[... 224 lines cut ...]
[ 95%] Building C object zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj
[ 95%] Linking C static library liboutput_lib.a
[ 96%] Built target output_lib
Scanning dependencies of target output_obj_renamed
[ 96%] Generating kobject_hash_renamed.o
[ 96%] Built target output_obj_renamed
Scanning dependencies of target linker_pass_final_script_target
[ 97%] Generating linker_pass_final.cmd
[ 97%] Built target linker_pass_final_script_target
[ 97%] Generating isr_tables.c
Scanning dependencies of target zephyr_final
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[100%] Linking C executable zephyr.elf
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr.elf section `priv_stacks_noinit' will not fit in region `SRAM'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 320 bytes
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_final.dir/build.make:120: recipe for target 'zephyr/zephyr.elf' failed
make[2]: *** [zephyr/zephyr.elf] Error 1
make[1]: *** [zephyr/CMakeFiles/zephyr_final.dir/all] Error 2
CMakeFiles/Makefile2:1364: recipe for target 'zephyr/CMakeFiles/zephyr_final.dir/all' failed
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
script returned exit code 2
```

ioannisg, 2019-08-19:
@dleach02 will you have time to address this one?

dleach02, 2019-08-19:
I will look at this today

dleach02, 2019-08-19:
These test cases are not valid for the current rt1015 platform as their min_ram requirements exceed the current RAM sizing of the rt1015. I suspect the test cases were not filtered correctly by @hakehuang.

dleach02, 2019-08-19:
I'm closing this bug report as it appears our internal testing is not properly filtering test cases were our target board does not meet minimum RAM requirements.

closed 2019-08-19 by dleach02