zephyrproject-rtos/zephyr #18284: tests/kernel/fp_sharing/float_disable fails on twr_ke18f

hakehuang, 2019-08-15:
**Describe the bug**
meet same memory region issue with user mode enalbed

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
PASS

**Impact**
float point usage

**Screenshots or console output**
```
***** Booting Zephyr OS version 2.0.0-rc1 *****

ASSERTION FAIL [num_parts <= max_partitions] @ ZEPHYR_BASE/kernel/mem_domain.c:90

	

FATAL: r0/a1:  0x00000004  r1/a2:  0x0000005a  r2/a3:  0x00800000

FATAL: r3/a4:  0x00000000 r12/ip:  0x00000008 r14/lr:  0x00006ea3

FATAL:  xpsr:  0x61000000

FATAL: s[0]:  0x00000000  s[1]:  0x00000000  s[2]:  0x00000000  s[3]:  0x00000000

FATAL: s[4]:  0x00000000  s[5]:  0x00000000  s[6]:  0x00000000  s[7]:  0x00000000

FATAL: s[8]:  0x00000000  s[9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000

FATAL: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000

FATAL: fpscr:  0x00000000

FATAL: Faulting instruction address (r15/pc): 0x00009a96

FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic

FATAL: Current thread: 0x20000454 (main)

FATAL: Halting system

```
**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - R2.0.0-rc1

hakehuang, 2019-08-15:
@MaureenHelm

ioannisg, 2019-08-16:
That's a known issue - we need to do something for this board that does not have enough MPU regions for basic user mode tests/samples. The tests passes for other platforms with NXP_MPU, so that's a low priority bug.

PR #18647 (2019-09-03): Test fix ke1xf userspace failures
Add an option to exlude HW Stack protection from user mode tests. This allows us to run the user mode test suite for platforms with limited number of MPU regions, such as the twr_ke18f board.
Fixes #18284 
Fixes #16229
Fixes #16225
closed 2019-09-03 by ioannisg