zephyrproject-rtos/zephyr #18297: Bluetooth: SMP: Pairing issues

cedricbois, 2019-08-15:
**Describe the bug**
When connecting, pairing and disconnecting with a peripheral, several times in a row, after some connections the pairing does fail.

The failure does happen on hardware (two nrf52840-DK) and in babbelsim simulation.
On babbelsim the thrown error is : SMP Timeout.
On the two nrf52840-DK the error is: the pairing_failed callback is triggered.

**To Reproduce**
Steps to reproduce the behavior:
1. connect to peripheral
2. start pairing using just works pairing :  bt_conn_security(pConnection, BT_SECURITY_MEDIUM);
3. See the pairing_confirmed callback triggered
4. disconnect the connection to the peripheral
5. repeate from step 1.
6. After about 5 to 40 successful pairings the pairing does fail.

**Expected behavior**
The loop should work indefinitely, especially in the babbelsim simulation which isn't sensitive to interferences.

**Impact**
System has no deterministic behaviour. Reliability is compromised.

**Screenshots or console output**
BabbelSim output: (Failure happens at the 21th pairing request).
```
...
 @00:00:20.071242   loop [21]
d_00: @00:00:20.071242  connecting to device
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_set_state: disconnected -> connect-scan
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_ref: handle 0 ref 3
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.071242  [00:00:20.071,228] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.073505  [00:00:20.073,486] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.073505  [00:00:20.073,486] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.073753  [00:00:20.073,730] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.073753  [00:00:20.073,730] <dbg> bt_conn.bt_conn_set_state: connect-scan -> connect
d_00: @00:00:20.073753  [00:00:20.073,730] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_set_state: connect -> connected
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x001b8820 to poll list
d_00: @00:00:20.101619  
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_le_param_update: conn 0x001b8820 features 0x00 params (18-28 0 400)
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.send_conn_le_param_update: conn 0x001b8820 features 0x00 params (18-28 0 400)
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x001b8820 to poll list
d_00: @00:00:20.101619  [00:00:20.101,593] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.104440  [00:00:20.104,431] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.104440  [00:00:20.104,431] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.104440  [00:00:20.104,431] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.104440  [00:00:20.104,431] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x001b8820 to poll list
d_00: @00:00:20.551879  [00:00:20.551,879] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.551879  [00:00:20.551,879] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
d_00: @00:00:20.551879  [00:00:20.551,879] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_send_cb: conn handle 0 buf len 11 cb 0x00000000 user_data 0x00000000
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_process_tx: conn 0x001b8820
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.send_buf: conn 0x001b8820 buf 0x001ac35c len 11
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.send_frag: conn 0x001b8820 buf 0x001ac35c len 11 flags 0x00
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_notify_tx: conn 0x001b8820
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.add_pending_tx: conn 0x001b8820 cb 0x00000000 user_data 0x00000000
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_ref: handle 0 ref 3
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_prepare_events: 
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x001b8820 to poll list
d_00: @00:00:20.553087  [00:00:20.553,070] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2
d_00: @00:00:20.553087  Security Level used is: 2
[... 341 lines cut ...]
[09:57:57:021] [00:01:07.111,999] <0x1b>[0m<dbg> bt_conn.bt_conn_set_state: connected -> disconnected<0x1b>[0m␍␊
[09:57:57:031] [00:01:07.112,121] <0x1b>[0m<dbg> bt_conn.bt_conn_unref: handle 0 ref 1<0x1b>[0m␍␊
[09:57:57:034] [00:01:07.112,182] <0x1b>[0m<dbg> bt_conn.bt_conn_prepare_events: <0x1b>[0m␍␊
[09:57:57:043] [00:01:07.112,182] <0x1b>[0m<dbg> bt_conn.bt_conn_notify_tx: conn 0x200013a4<0x1b>[0m␍␊
[09:57:57:047] [00:01:07.112,182] <0x1b>[0m<dbg> bt_conn.bt_conn_unref: handle 0 ref 0<0x1b>[0m␍␊
```
**Log on Hardware, peripheral side:**

```
[loop 21]
[09:57:07:427] [CONN] connected. Active connections: 1.␍␊
[09:57:38:338] PAIRING FAILED ␍␊
[09:57:52:575] [CONN] disconnected (reason 34 /0x22). Active connections: 0.␍␊
```

**Environment (please complete the following information):**
 - OS: Kubuntu
 - Toolchain is Zephyr SDK 0.10.1
 - Commit SHA or Version used
3bddc20f87f2aa28ade499689532f6be9f7ee865

**Additional context**
Please let me know if any additional information is needed.

Thank you for your help.

joerchan, 2019-08-16:
I'm seeing something similar here. Will investigate.

joerchan, 2019-08-22:
@cedricbois We believe we have found the problem. Can you confirm that https://github.com/zephyrproject-rtos/zephyr/pull/18569 fixes your stability issue?

carlescufi, 2019-08-22:
@cedricbois can you please test again once we have merged:

https://github.com/zephyrproject-rtos/zephyr/pull/18569
and
https://github.com/zephyrproject-rtos/zephyr/pull/18578

cedricbois, 2019-08-23:
Yes I will do the tests, but I will only be able to test the fixes on monday 26/08. Sorry for the delay.

ioannisg, 2019-08-26:
@cedricbois the above PRs are not merged, so you can test latest master.

carlescufi, 2019-08-27:
@cedricbois can you please retest and confirm that the issue is fixed?

joerchan, 2019-08-30:
Tested and reproduced this issue on v1.14-branch.
Connected using zephyr on both side, with modified central and peripheral sample.

Both:
 - Disabled privacy
 - Enabled Legacy LL
 - Enabled Debug SMP keys.
 - Set CONN_UPDATE timeout to 450 ms. (this makes it timeout during the encryption procedure).

Central:
 - Connect to peripheral in scan report using direct address.
 - In connected callback initiate security with level 2
 - In conn param update callback disconnect link.

Peripheral
 - No other changes

Failure can happen in 2 ways.

1. Encrypt procedure fails because master sends a packet with invalid MIC
Leads to disconnect with error code 0x3d on peripheral side, master disconnects with reason 0x08 (supervision timeout when slave drops link).
2. Encrypt procedure times out because slave does not send LL_ENC_RSP. 
Leads to disconnect with error code 0x22 (LL procedure timeout) on peripheral side, SMP will timeout before this, because SMP timeout is shorter. 
Master side seems to have disconnected with error code 0x08, connection timeout. This looks wierd as the master should have dropped the link at procedure timeout, but instead ran to supervision timeout.

joerchan, 2019-09-02:
[smp_pairing_issues.zip](https://github.com/zephyrproject-rtos/zephyr/files/3564961/smp_pairing_issues.zip)
Added sniffer logs and console output.
Sniffer log (mic failure): Issue happens at 24678
Sniffer log (smp timeout): Issue happens at 43179

joerchan, 2019-09-02:
Reproduced SMP timeout / LL control procedure timeout issue on Legacy controller in master branch (47f36afdb9).

joerchan, 2019-09-02:
Sniffer trace from reporter shows that the CONN UPDATE is not colliding with the Encryption procedure, so the point "Set CONN_UPDATE timeout to 450 ms. (this makes it timeout during the encryption procedure)." appears to not be a trigger for the bug.
In this case packets with MIC errors was sent by the master link with no apparent LL control procedure collision, and no other data other than the SMP pairing packets.

joerchan, 2019-09-02:
So far unable to reproduce this with the Split controller.

cvinayak, 2019-09-05:
> 
> 
> [smp_pairing_issues.zip](https://github.com/zephyrproject-rtos/zephyr/files/3564961/smp_pairing_issues.zip)
> Added sniffer logs and console output.
> Sniffer log (mic failure): Issue happens at 24678
> Sniffer log (smp timeout): Issue happens at 43179
> Diff file to reproduce using central and peripheral samples.

At 24668. LL_ENC_REQ is sent with SN: 0. NESN: 0.
At 24670. Empty PDU is sent with SN:0. NESN: 0.

Which implies the LL_ENC_REQ was not really transmitted/acknowledge when it comes central.
The consequence being that controller is not aware of LL_ENC_REQ ever being acknowledged by the peripheral, hence central thinks to be not in encryption setup. A LL_CONNECTION_PARAM_REQ is (probably) responded with LL_CONNECTION_UPDATE_REQ in 24678 which is detected by peripheral as MIC failure.

This relates to dropped LL_ENC_REQ by the master on retransmission which is fixed in https://github.com/zephyrproject-rtos/zephyr/pull/15320/commits/31256568a2835e7f43f7a627203ddb0ae2d74563 as part of #15320

I conclude that commits in #15320 cover the issues reported in this issue.

cvinayak, 2019-09-05:
> 
> 
> So far unable to reproduce this with the Split controller.

The fact that PDUs are not retransmitted between ULL and LLL does not exposed the bug in `is_enc_req_pause_tx` . It is the retransmission that leads to calling `is_enc_req_pause_tx` mutate the states in the v1.14 controller.

joerchan, 2019-09-05:
@cvinayak That looks like the correct analysis.

PR #15320 (2019-09-07): Bluetooth: controller: Fix control procedure sequencing
Fix implementation to correctly cache the control procedures
initiatable by local and peer. And, fix feature exchange and
version information procedures from being disallowed by
having then as cached requests to the controller.

Relates to #15256.
Fixes #18297.
closed 2019-09-07 by carlescufi