zephyrproject-rtos/zephyr #18298: Unable to build mesh-demo for BBC micro:bit

dallonby, 2019-08-15:
---
name: Unable to build mesh-demo for BBC micro:bit
about: SRAM exceeded by 64 bytes
title: ''
labels: bug
assignees: ''

---
**Bug**
Using both the GNU gcc toolchain, and the Zephyr 0.10.2 SDK, attempting to build this firmware fails with the following:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       83866 B       256 KB     31.99%
            SRAM:       16448 B        16 KB    100.39%
        IDT_LIST:         136 B         2 KB      6.64%
/home/david/zephyr-sdk-0.10.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr_prebuilt.elf section `_net_buf_pool_area' will not fit in region `SRAM'
/home/david/zephyr-sdk-0.10.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000020004000,0000000020004087] overlaps section _net_buf_pool_area VMA [0000000020003fc0,000000002000403f]
/home/david/zephyr-sdk-0.10.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 64 bytes
collect2: error: ld returned 1 exit status```

Tested using both Ninja and make.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=bbc_microbit ..
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu
 - Toolchain Zephyr SDK / GNU make
 - Commit 5d30d49

carlescufi, 2019-08-15:
@dallonby see https://github.com/zephyrproject-rtos/zephyr/pull/18294

dallonby, 2019-08-15:
thanks, confirmed fixed.

closed 2019-08-15 by carlescufi

WilliamGFish, 2019-08-15:
@carlescufi 
Hi,
I just moved to v2.0 rc1 and built my app on the M-Bit and getting hard fault. So I turned to the Mesh Demo to test and double check changes, unfortunately if I set the address to 0x000F therefore invoking Heartbeat via Health Server it crashes often on boot.

Sometimes after a reset it will last a few seconds which may correspond to a heartbeat message.

I tried changing buffer sizes but no luck. Sorry but debugging no easy on these things

carlescufi, 2019-08-16:
@WilliamGFish 
can you try obtaining a backtrace or an `addr2line` on those crashes? We need a bit more info to reproduce. Maybe @jhedberg can give the mesh demo a go on the micro:bit.

WilliamGFish, 2019-08-16:
Okay, I'll look over the weekend.

WilliamGFish, 2019-08-19:
Ok, Attempted to trace (using windows so harder) used OZone to do reverse lookup on following error:
```
Bluetooth initialized
Mesh initialized
Loading stored settings
Using stored settings
Subscribing to heartbeat messages
FATAL: ***** HARD FAULT *****
FATAL: r0/a1:  0x20000bb4  r1/a2:  0x00000000  r2/a3:  0x00000002
FATAL: r3/a4:  0x000122cd r12/ip:  0x20000bb4 r14/lr:  0x000096af
FATAL:  xpsr:  0x00000000
FATAL: Faulting instruction address (r15/pc): 0x000000fe
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x20000e48 (unknown)
FATAL: Halting system
```

```
__aeabi_uidiv
__udivsi3
$Thumb
 000000D0   MOVS         R2, #0
 000000D2   LSRS         R3, R0, #1
 000000D4   CMP          R3, R1
 000000D6   BCC          0x000001C2
 000000D8   LSRS         R3, R0, #4
 000000DA   CMP          R3, R1
 000000DC   BCC          0x0000019E
 000000DE   LSRS         R3, R0, #8
 000000E0   CMP          R3, R1
 000000E2   BCC          0x0000016E
 000000E4   LSRS         R3, R0, #12
 000000E6   CMP          R3, R1
 000000E8   BCC          0x0000013C
 000000EA   LSRS         R3, R0, #16
 000000EC   CMP          R3, R1
 000000EE   BCC          0x0000010C
 000000F0   MOVS         R2, #0xFF
 000000F2   LSLS         R1, R1, #8
 000000F4   REV          R2, R2
 000000F6   LSRS         R3, R0, #16
 000000F8   CMP          R3, R1
 000000FA   BCC          0x00000102
 000000FC   ASRS         R2, R2, #8
 **000000FE   LSLS         R1, R1, #8**
```

If i look at the zephyr.lst file i can see this:

>       fa:	d302      	bcc.n	102 <CONFIG_FLASH_SIZE+0x2>
>       fc:	1212      	asrs	r2, r2, #8
>       fe:	0209      	lsls	r1, r1, #8

Hopefully this is of some use.

WilliamGFish, 2019-08-19:
@carlescufi 
Update:

```
$ addr2line -e zephyr.elf 0x000000fe
??:?

```
Other intermittent errors

```
$ addr2line -e zephyr.elf 0x20000b1c
C:/Users/BillyBob/zephyrGoodX/zephyrNew/subsys/bluetooth/controller/ll_sw/ull.c:111

$ addr2line -e zephyr.elf 0x200003bc
C:/Users/BillyBob/zephyrGoodX/zephyrNew/subsys/bluetooth/controller/ticker/ticker.c:223

$ addr2line -e zephyr.elf 0x00000002
C:/Users/BillyBob/zephyrGoodX/zephyrNew/lib/os/fdtable.c:116

```

carlescufi, 2019-08-21:
@WilliamGFish thank you very much for the report. We are still debugging issues with the new Link Layer, and so I will ask you for some patience until we have fixed those and can then start analyzing further issues that might not be related to the current issue set.
@cvinayak is in charge of this process and we will update as appropriate. Our target is to have all those bug fixed by the release of 2.0.

carlescufi, 2019-08-22:
@WilliamGFish 

We have fixed a couple of critical bugs:

https://github.com/zephyrproject-rtos/zephyr/pull/18569
https://github.com/zephyrproject-rtos/zephyr/pull/18578

Could  you test again once both are merged?

>So I turned to the Mesh Demo to test and double check changes, unfortunately if I set the address to 0x000F therefore invoking Heartbeat via Health Server it crashes often on boot.

This sounds like it's unrelated to Bluetooth. Maybe @jhedberg can take a quick look since it's the mesh demo on micro:bit.

WilliamGFish, 2019-08-29:
I've fetched the changes and now consistently fails at ull.c:111

This can be generated by using button B repeatedly post boot without being the Heartbeat server. Often only takes 5-6 presses.

I have tried all sorts of different memory configurations but little effect. Even Logging a DBG level yielded little useful info.

I then used other BLE Mesh programs and they too fail at ull.c:111

PR #18294 (2019-08-15): Bluetooth fixes for build failures
Fix the Bluetooth samples in #18246

Fixes #18298