zephyrproject-rtos/zephyr #18301: menuconfig target can corrupt build configuration

pabigot, 2019-08-15:
#18261 is not a bug in the kernel, but a bug in the build system.

I ran this sequence:

```
rm -rf va && west build -d va -b nrf52840_pca10056
rm -rf vb && west build -d vb -b nrf52840_pca10056
west build -d vb -t menuconfig # disable TIMESLICING and save
west build -d vb 
mv prj.conf prj.conf-
echo 'CONFIG_TIMESLICING=n' >> prj.conf
rm -rf vc && west build -d vc -b nrf52840_pca10056
mv prj.conf- prj.conf
```

Variant `vb` produces the usage fault described in #18261, while variants `va` and `vc` work as expected.

A recursive diff of `v{b,c}` showed the resulting `zephyr/.config` was identical, but this file was not:
```
diff -ur vb/zephyr/include/generated/offsets.h vc/zephyr/include/generated/offsets.h
--- vb/zephyr/include/generated/offsets.h       2019-08-15 06:30:31.507371588 -0500
+++ vc/zephyr/include/generated/offsets.h       2019-08-15 06:30:51.943293720 -0500
@@ -14,8 +14,8 @@
 #define ___cpu_t_current_OFFSET 0x8
 #define ___cpu_t_nested_OFFSET 0x0
 #define ___cpu_t_irq_stack_OFFSET 0x4
-#define ___kernel_t_idle_OFFSET 0x20
-#define ___kernel_t_ready_q_OFFSET 0x24
+#define ___kernel_t_idle_OFFSET 0x1c
+#define ___kernel_t_ready_q_OFFSET 0x20
 #define ___ready_q_t_cache_OFFSET 0x0
 #define ___thread_base_t_user_options_OFFSET 0xc
 #define ___thread_base_t_thread_state_OFFSET 0xd
```

Inspection confirms that `include/generated/offsets.h` is not regenerated after the change to the application `.config` file, so the size of `struct _cpu` changes relative to what what it was before the Kconfig symbol set was changed.

`${OFFSETS_H_TARGET}` should have a dependency on `${DOTCONFIG}` to prevent these inconsistencies.  My cmake skills are not up to the task.

pabigot, 2019-08-15:
@carlescufi @pizi-nordic fyi

carlescufi, 2019-08-15:
@SebastianBoe could you please take a look  at this one?

PR #18304 (2019-08-15): cmake: Fix dependency between offset.c.obj and offsets.h
The command for generating offsets.h has only been depending on the
target for offsets.c.obj, and not the file offsets.c.obj itself. Both
are needed, as explained in "CMake: dependencies between targets and
files and custom commands"[0].

This patch adds the dependency and fixes #18301

[0] https://tinyurl.com/y2hgzjhx
commit 5962aab4 (2019-08-15): cmake: Fix dependency between offset.c.obj and offsets.h

The command for generating offsets.h has only been depending on the
target for offsets.c.obj, and not the file offsets.c.obj itself. Both
are needed, as explained in "CMake: dependencies between targets and
files and custom commands"[0].

This patch adds the dependency and fixes #18301

[0] https://tinyurl.com/y2hgzjhx
closed 2019-08-15 by carlescufi