zephyrproject-rtos/zephyr #18305: Native Posix target can not use features with newlib dependencies

vanwinkeljan, 2019-08-15:
**Describe the bug**
In case you are building for native posix target the newlib C library can not be selected but some features have a hard dependency towards newlib even when provided by OS libc implementation.

**To Reproduce**
One example more issues can be found in code base:
1. west build -b native_posix samples/hello_world -- -DCONFIG_CPLUSPLUS=y -DCONFIG_EXCEPTIONS=y

```
warning: EXCEPTIONS (defined at subsys/cpp/Kconfig:55) was assigned the value 'y' but got the value
'n'. You can check symbol information (including dependencies) in the 'menuconfig' interface (see
the Application Development Primer section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_EXCEPTIONS.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.
```

**Expected behavior**
You should be able to select all features that only depends on newlib for the native posix target.

**Impact**
Some features can not be enabled on native posix target

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc 7.4
 - Commit SHA: 4ea3c9560ac3823ea4a5f1a79e4661a499941b02

**Proposed Solution**

Add hidden Kconfig symbol `HAS_FULL_LIBC`, which is set both for native posix and newlib, and use this symbol in `depends on`.

**ALternative Solution**

Add `ARCH_POSIX` symbol to all `depends on` where newlib is used

vanwinkeljan, 2019-08-15:
@carlescufi @aescolar in case the proposed solution looks acceptable I could also take this one up

carlescufi, 2019-08-15:
Maybe @galak and @aescolar can comment on the suitability of the proposed solution

aescolar, 2019-08-15:
@vanwinkeljan The issue is, as you state, not in native_posix, but on setting `depends on NEWLIB_LIBC` as that assumes there is only 2 C libraries which can be used with Zephyr. This problem will exist for anybody who tries to use a 3rd C library, not just when targeting native_posix.

So, the alternative solution, of adding a `depends on ARCH_POSIX || NEWLIB_LIBC` on those features would not be a good solution, as that just hides the real problem in the tree.

I would think it may make more sense to do add a new `MINIMAL_LIBC` kconfig option, which `default y if !NEWLIB_LIBC`, and have the `depends on NEWLIB_LIBC` be instead `depends on !MINIMAL_LIBC`. Also have the lib/libc/CMakeLists.txt include minimal libc only if MINIMAL_LIBC instead of if not NEWLIB_LIBC

nashif, 2019-08-15:
related https://github.com/zephyrproject-rtos/zephyr/pull/18224

vanwinkeljan, 2019-08-15:
Pushed a PR (#18315) that adds a choice for c library selection and updated dependencies to `!MINIMAL_LIBC`

PR #18315 (2019-11-01): Added Kconfig choice for c library impl. + corrected newlib dependencies 
Added a Kconfig choice to select c library implementation and updated the dependencies towards newlib through the code base.

Fixes #18305
closed 2019-11-01 by galak