zephyrproject-rtos/zephyr #18306: Unable to reconnect paired devices with controller privacy disabled (host privacy enabled)

mtpr-ot, 2019-08-15:
**Description**
With host privacy enabled and controller privacy disabled, it is not possible to re-connect to a paired device.

**To Reproduce**
_Preconditions:_
CONFIG_BT_PRIVACY=y
CONFIG_BT_CTLR_PRIVACY=n

Peripheral advertises constantly. Central scans, connects, disconnects and then scans and finally attempts to re-connect. At second attempt:
- Scanner scans for ADV packets, sees our peripheral by random address and sends scan report
- In the host layer, the advertiser address is looked up and a key is found
- The random address is translated to public address and app is notified
- The app takes the public address of the advertiser and initiates a connection
- The scanner receives connection information containing the public address
- Scanner waits forever for a match between received random address and the public address from app

**Expected behavior**
Central and peripheral should connect every time.

**Impact**
Unable to connect after devices have paired.

**Environment (please complete the following information)**
 - OS: proprietary
 - Toolchain: proprietary
 - Zephyr 2.0 RC1

aescolar, 2019-08-16:
@joerchan This seems like a regression. But please remove the label if it is not.

joerchan, 2019-08-16:
Yes, regression from here: 45da629b24f720019355fca8e1ddddb798ac409f

PR #18329 (2019-08-16): Bluetooth: Host: Fix unable to connect using host resolution
Fix issue unable to connect to bonded peer when host resolution is used
either because the controller does not support privacy, or the
controller resolving list was exceeded.
In this case we need to use the RPA from the advertising report
directly, there is a small chance of the peer cycling the RPA here, in
which case the connection might be unsuccessful.

Bug introduced here: 45da629b24f720019355fca8e1ddddb798ac409f

Fixes: #18306
commit 8ebb117e (2019-08-16): Bluetooth: Host: Fix unable to connect using host resolution

Fix issue unable to connect to bonded peer when host resolution is used
either because the controller does not support privacy, or the
controller resolving list was exceeded.
In this case we need to use the RPA from the advertising report
directly, there is a small chance of the peer cycling the RPA here, in
which case the connection might be unsuccessful.

Bug introduced here: 45da629b24f720019355fca8e1ddddb798ac409f

Fixes: #18306
commit 58264518 (2019-08-16): Bluetooth: Host: Fix unable to connect using host resolution

Fix issue unable to connect to bonded peer when host resolution is used
either because the controller does not support privacy, or the
controller resolving list was exceeded.
In this case we need to use the RPA from the advertising report
directly, there is a small chance of the peer cycling the RPA here, in
which case the connection might be unsuccessful.

Bug introduced here: 45da629b24f720019355fca8e1ddddb798ac409f

Fixes: #18306
commit c19dac1a (2019-08-16): Bluetooth: Host: Fix unable to connect using host resolution

Fix issue unable to connect to bonded peer when host resolution is used
either because the controller does not support privacy, or the
controller resolving list was exceeded.
In this case we need to use the RPA from the advertising report
directly, there is a small chance of the peer cycling the RPA here, in
which case the connection might be unsuccessful.

Bug introduced here: 45da629b24f720019355fca8e1ddddb798ac409f

Fixes: #18306
closed 2019-08-16 by jhedberg