zephyrproject-rtos/zephyr #18320: tests/drivers/can/api/peripheral.can fails on FRDM-K64F

galak, 2019-08-15:
**Describe the bug**
Test timesout in `test_set_loopback`

This is all the output we get:
```
***** Booting Zephyr OS build v2.0.0-rc1-63-g25a9cde22349 *****
Running test suite can_driver
===================================================================
starting test - test_set_loopback
```

**To Reproduce**
`scripts/sanitycheck -p frdm_k64f --device-testing --device-serial /dev/ttyACM0 -s tests/drivers/can/api/peripheral.can`

galak, 2019-08-15:
The issue appears to be in `FLEXCAN_Enable()`.  When its called to disable the FLEXCAN, we spin for ever waiting on the hardware to go to low power mode:

```
while (!(base->MCR & CAN_MCR_LPMACK_MASK))
```

galak, 2019-08-15:
This also appears related to reset in some form.

alexanderwachter, 2019-08-16:
This issue can occur when the CAN0_RX pin is floating. The low-power-mode can only be switched when the controller is idling. The bus is in the recessive state (idling) when the RX pin is logical high.

PR #18328 (2019-08-16): boards: frdm_k64f: Enable pullup for CAN0_RX pin
This commit enables the pullup on CAN0_RX pin (PORTB 19).
The pullup ensures that the CAN controller initializes even
without a transceiver connected.

Fixes #18320.
closed 2019-08-16 by MaureenHelm