zephyrproject-rtos/zephyr #18326: Bluetooth: Mesh: LPN: Remove msg from cache on rejection Enhancement

menglingao1997, 2019-08-16:
5d5706e785275a160d6ec6416f69289d01db17b7
#17809

``` C
	if (IS_ENABLED(CONFIG_BT_MESH_LOW_POWER) &&
	    bt_mesh_lpn_established() && rx->net_if == BT_MESH_NET_IF_ADV &&
	    (!bt_mesh_lpn_waiting_update() || !rx->friend_cred)) {
		BT_WARN("Ignoring unexpected message in Low Power mode");
		return -EAGAIN;
	}
```
I think the best way to deal with this problem is to move this piece of code to**net_find_and_decrypt()**
Just like this:

```C
static bool net_find_and_decrypt(const u8_t *data, size_t data_len,
				 struct bt_mesh_net_rx *rx,
				 struct net_buf_simple *buf)
{
	struct bt_mesh_subnet *sub;
	int i;

	BT_DBG("");

	for (i = 0; i < ARRAY_SIZE(bt_mesh.sub); i++) {
		sub = &bt_mesh.sub[i];
		if (sub->net_idx == BT_MESH_KEY_UNUSED) {
			continue;
		}

#if (defined(CONFIG_BT_MESH_LOW_POWER) || \
     defined(CONFIG_BT_MESH_FRIEND))
		if (!friend_decrypt(sub, data, data_len, rx, buf)) {
			rx->friend_cred = 1U;
			rx->ctx.net_idx = sub->net_idx;
			rx->sub = sub;
			return true;
		}
#endif
               if (IS_ENABLED(CONFIG_BT_MESH_LOW_POWER) &&
	             bt_mesh_lpn_established() && rx->net_if == BT_MESH_NET_IF_ADV &&
	             (!bt_mesh_lpn_waiting_update() || !rx->friend_cred)) {
		            BT_WARN("Ignoring unexpected message in Low Power mode");
		     return -EAGAIN;
	       }

		if (NID(data) == sub->keys[0].nid &&
		    !net_decrypt(sub, sub->keys[0].enc, sub->keys[0].privacy,
				 data, data_len, rx, buf)) {
			rx->ctx.net_idx = sub->net_idx;
			rx->sub = sub;
			return true;
		}

		if (sub->kr_phase == BT_MESH_KR_NORMAL) {
			continue;
		}

		if (NID(data) == sub->keys[1].nid &&
		    !net_decrypt(sub, sub->keys[1].enc, sub->keys[1].privacy,
				 data, data_len, rx, buf)) {
			rx->new_key = 1U;
			rx->ctx.net_idx = sub->net_idx;
			rx->sub = sub;
			return true;
		}
	}

	return false;
}
```

menglingao1997, 2019-08-16:
@jhedberg

jhedberg, 2019-08-16:
@richardbarlow FYI. @menglingao1997 the main reason the check is in `transport.c` is that Friendship is a Transport-layer feature.

richardbarlow, 2019-08-16:
@jhedberg When I was fixing this one thing I was wondering was whether this complexity could be removed by only attempting to decrypt messages using the friend key when the LPN has an established friendship. I agree that the current separation of checking if a message should be rejected in the transport layer and the subsequent removal from it in the network layer is the correct approach, but it would be nice to avoid the erroneous messages ever being decrypted (and therefore ending up in the cache) in the first place.

Another advantage of this is that it will reduce the power consumption of LPNs slightly, as they will not needlessly attempt to decrypt every message with all of its non-friend keys.

I'm not sure if this approach violates any aspects of the specification. I don't think it should since an LPN must only process messages encrypted with the friend key.

menglingao1997, 2019-08-16:
@richardbarlow Whether the low-power node should accept the message from the non-friend relationship, the specification does not make a statement, I think it is enough to accept only the message from the friend, so that not only can the complexity of the low-power node be reduced, but the main purpose is to save power. Isn't it?

menglingao1997, 2019-08-16:
@jhedberg Appropriate layer crossing is not a bad thing, always insisting on the layer-to-layer separation mode, which will lead to some strange phenomena, just like I think now.

jhedberg, 2019-08-16:
@menglingao1997 @richardbarlow fair enough. Does one of you want to submit a PR? It should preferably have a code comment to explain the apparent layer violation.

richardbarlow, 2019-08-16:
@jhedberg To be clear, I don't condone violating the layering.

I was thinking along the lines of `net.c` having a concept of the currently active key set (for all subnets). When a node is *not* an LPN this is the normal set of keys, when a node is an LPN this is the set of friend keys. This way `net.c` doesn't need to know anything about 'friends' or 'LPNs' at all, it simply tries to decrypt messages with the active key set and pass them up to `transport.c` when successfully decrypted. I think this approach could make `net.c` simpler and actually improve the separation of the layers.

However, this is not as easy as I first thought as the friend key management that occurs in `net.c` is also used by the friend node itself, which possibly makes things more complicated. I suspect that first separating the key management out of `net.c` might provide a path towards simplifying this, but that is quite a large undertaking. Unfortunately I cannot put any time towards doing this now as the project that required BT Mesh is almost finished and other projects now have a higher priority.

In conclusion, I think that it should be left as it is. If anything is to be done to improve this I think that adding more separation of concerns (i.e. key management) will make it both more readable and more performant. I don't think there is much to gain from breaking the layering in this one specific case.

It's also worth noting that this issue has arisen due to lack of clarity in how this situation should be handled in the specification, rather than due to rigidly enforcing the layering. The ideal situation would be for the specification to be made more clear (so that all implementations benefit) rather than making Zephyr's implementation slightly mangled (with the layer violation).

menglingao1997, 2019-08-16:
@richardbarlow @jhedberg I respect everyone's point of view, and if there is a better solution in the future, it would be best.

jhedberg, 2019-08-16:
> I suspect that first separating the key management out of net.c might provide a path towards simplifying this, but that is quite a large undertaking. 

I've had this on my TODO for awhile, but just never got around to it. I'd move the code from `net.c` to a new `friend_cred.c`. I'd be happy to review such a PR if anyone has time to look into it.

Anyway, I'm fine with the consensus of deferring this optimisation to some point in the future.

closed 2019-08-16 by menglingao1997