zephyrproject-rtos/zephyr #18351: logging: 32 bit float values don't work.

deadsy, 2019-08-17:
**Code**

```
float x = 1.234f;
printf("this is a 32 bit float %f\n", x);
LOG_INF("this is a 32 bit float %f", x);
```

**Output**

> ***** Booting Zephyr OS build v2.0.0-rc1-50-g70640334730b *****
> ....
> this is a 32 bit float 1.234000
> [00:00:00.008,000] <inf> ggm.main: this is a 32 bit float 0.000000

**Notes**
CONFIG_LOG_ENABLE_FANCY_OUTPUT_FORMATTING=y
This uses the minimal libc print formatter for the log messages. The print formatter works for float formatting when invoked via printf, suggesting that the float value is somehow being clobbered in the log functions.

**Environment**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-8-2018-q4-major
 - Board: mimxrt1020_evk

pabigot, 2019-08-17:
A summary of the analysis only; I'd prefer @nordic-krch take on this task as it appears to be complex.

The path this code takes through the logging system involves counting the number of arguments and caching one `log_arg_t` value for each one in an array, then reconstructing the original call stack in the output context.

The count is a simple scan for the number of conversion specifiers.  It does not take into account length modifiers or the specifier character itself, and so is incorrect when any given argument exceeds the size of `log_arg_t` on the target, as is the case with `%f` and `%llu` on 32-bit hosts.  Any conversions that follow the oversized conversion will also be incorrect.

A fix requires an analysis of the format string to determine the number of `log_arg_t` values required for each conversion.  This analysis will be specific to the selected back-end formatter (either `z_prf()` or `z_vprintk()`) and any Kconfig symbols that control the behavior of that formatter.

dleach02, 2019-12-10:
Removing 2.1 milestone

koffes, 2020-10-11:
This is still an issue.
The following example has been compiled with:
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
CONFIG_NEWLIB_LIBC=y

```
float32_t val = 1.2345;
char buf[25];
sprintf(buf, "sprintf %f", val);
LOG_WRN("%s", log_strdup(buf)); //Yields: "sprintf 1.234500"
printk("printk %f\n", val); //Yields: "printk %f"
printf("printf %f\n", val); //Yields: "printf 1.234500" 
LOG_WRN("LOG_WRN %f", val); //Yields: "LOG_WRN %f"
```
If these configs are enabled the system should be able to print floating point numbers regardless of print method, If this is not feasible/desirable, there should be some ASSERT or warning when using %f or %e.

koffes, 2020-10-11:
@carlescufi can this issue be reopened?

pabigot, 2020-10-11:
Yes.  It shouldn't have been closed.

pabigot, 2020-10-27:
> The count is a simple scan for the number of conversion specifiers. It does not take into account length modifiers or the specifier character itself, and so is incorrect when any given argument exceeds the size of `log_arg_t`

This is something that might be addressed if we build a fully-functional Zephyr API for printf-style formatting with stream output, which is the subject of https://github.com/zephyrproject-rtos/zephyr/pull/29203#issuecomment-710072323.

pabigot, 2020-11-16:
With [cbprintf_arglen](https://docs.zephyrproject.org/latest/reference/misc/formatted_output.html#c.cbprintf_arglen) it should now be possible to fix this and all the other bugs related to passing 64-bit values for log formatting.  @nordic-krch @carlescufi

nordic-krch, 2020-11-17:
@pabigot noted, i will take a look at that.

deadsy, 2021-01-17:
I'm running zephyr 2.4.0.
It still doesn't work.

nordic-krch, 2021-01-18:
yes, it will be addressed soon, see #30353.

closed 2021-04-19 by nashif

nordic-krch, 2021-04-20:
floats can now be logged with new logging. Currently legacy version is the default one, so it has to be explicitly enabled (`CONFIG_LOG2_MODE_DEFERRED`).

manoj153, 2021-05-24:
> floats can now be logged with new logging. Currently legacy version is the default one, so it has to be explicitly enabled (`CONFIG_LOG2_MODE_DEFERRED`).

hi  have tried setting `CONFIG_LOG2_MODE_DEFERRED=y` with the latest zephyr build and yet to print float types, 
`%f`, is just printing out as `"%f"` , instead of given variable's value

nordic-krch, 2021-05-25:
@manoj153 please check if float formatting is enabled in cbprintf `CBPRINTF_FP_SUPPORT`.

manoj153, 2021-08-02:
> CBPRINTF_FP_SUPPORT

yes that helps. works now