zephyrproject-rtos/zephyr #18352: devicetree: support multiple values in io-channels

pabigot, 2019-08-17:
The `io-channels` property was added in #18101 but the implementation doesn't seem to support multiple channels, although edtlib internal documentation suggests it should.

For example, I should be able to define a bank of ADC inputs corresponding to thermistors with something like
```
thermistor-bank {
	compatible = "voltage-divider";
	io-channels = <&adc 1>, <&adc 2>, <&adc 4>;
	output-ohms = <10000>;
};
```
Currently this produces multiple definitions for the same symbol.  I also tried:
```
io-channels = <&adc 1 &adc 2 &adc 4>;
```
with the same result, though edtlib seems to handle something similar for `foo-gpios`.

See [this example](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts?h=v5.3-rc4#n111).

ulfalizer, 2019-08-19:
@pabigot 
Could check what `write_iochannels()` does in `gen_defines.py`, and compare it against `write_gpios()`. Honestly, I don't know what output the code expects for all of these, so I might not be the best person to change it. :P

It's made to be easy to tweak at least. `gen_defines.py` decides what the output is, while edtlib is general.

ulfalizer, 2019-08-19:
Maybe @jimparis has opinions.

ulfalizer, 2019-08-19:
Re. `io-channels = <&adc 1>, <&adc 2>, <&adc 4>;` vs. `io-channels = <&adc 1 &adc 2 &adc 4>;`, they're treated exactly the same, and generate the same output with the devicetree compiler as well.

I added support for the first form in the type checker even though it's redundant, because it felt a bit mean to disallow something people probably do because they care about trying to make things readable.

PR #18498 (2019-08-19): scripts/dts/gen_defines.py: Fix generation for multiple IO Channels
If there is more than one IO Channel than generate a define with a
trailing index for the IO Channel.  This matches what we do for GPIOs
and PWMs.

So something like:
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_0
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_1
  ...
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_<N>

Fixes #18352
commit a0a576d3 (2019-08-19): scripts/dts/gen_defines.py: Fix generation for multiple IO Channels

If there is more than one IO Channel than generate a define with a
trailing index for the IO Channel.  This matches what we do for GPIOs
and PWMs.

So something like:
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_0
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_1
  ...
  DT_FOOBAR_IO_CHANNELS_CONTROLLER_<N>

Fixes #18352
closed 2019-08-19 by MaureenHelm