zephyrproject-rtos/zephyr #18355: [Coverity CID :203466]Memory - illegal accesses in /drivers/can/can_mcux_flexcan.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /drivers/can/can_mcux_flexcan.c
 Category: Memory - illegal accesses
 Function: mcux_flexcan_send
 Component: Drivers
 CID: 203466
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

alexanderwachter, 2019-08-24:
CID 203466 (#4 of 4): Out-of-bounds read (OVERRUN)10. overrun-local: Overrunning array data->tx_cbs of 11 52-byte elements at element index 11 (byte offset 623) using index alloc (which evaluates to 11).

Line 218

henrikbrixandersen, 2019-08-24:
That should be safe to ignore. The counting semaphore will ensure that this can never happen.

alexanderwachter, 2019-08-24:
Set coverty state to „ignore“

closed 2019-08-24 by alexanderwachter