zephyrproject-rtos/zephyr #18359: [Coverity CID :203521]Memory - illegal accesses in /drivers/espi/espi_mchp_xec.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /drivers/espi/espi_mchp_xec.c
 Category: Memory - illegal accesses
 Function: espi_xec_receive_vwire
 Component: Drivers
 CID: 203521
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203521:  Memory - illegal accesses  (OVERRUN)
/drivers/espi/espi_mchp_xec.c: 299 in espi_xec_receive_vwire()
293     	    (xec_id >= ESPI_SMVW_IDX_MAX) || (level == NULL)) {
294     		return -EINVAL;
295     	}
296     
297     	if (signal_info.dir == ESPI_MASTER_TO_SLAVE) {
298     		ESPI_MSVW_REG *reg = &(ESPI_M2S_VW_REGS->MSVW00) + xec_id;
>>>     CID 203521:  Memory - illegal accesses  (OVERRUN)
>>>     Overrunning array of 12 bytes at byte offset 108 by dereferencing pointer "reg".
299     		*level = ((reg->SRC >> (src_id << 3)) & 0x01ul);
300     	}
301     
302     	if (signal_info.dir == ESPI_SLAVE_TO_MASTER) {
303     		ESPI_SMVW_REG *reg = &(ESPI_S2M_VW_REGS->SMVW00) + xec_id;
304     		*level = ((reg->SRC >> (src_id << 3)) & 0x01ul);
```

galak, 2019-08-19:
Not a bug, this is due to how register access/pointer math is working.

PR #18493 (2019-08-19): drivers: espi: espi_mchp_xec: Fix wrong size type for status var
In status variable in espi_pc_isr should have been a 32-bit unsigned int
as ESPI_PC_REGS->PC_STATUS is 32-bits.

Fixes #18359
Coverity-CID: 203521
commit 59bbf3a5 (2019-08-19): drivers: espi: espi_mchp_xec: Fix wrong size type for status var

In status variable in espi_pc_isr should have been a 32-bit unsigned int
as ESPI_PC_REGS->PC_STATUS is 32-bits.

Fixes #18359
Coverity-CID: 203521
closed 2019-08-19 by galak