zephyrproject-rtos/zephyr #18361: [Coverity CID :203515]Uninitialized variables in /drivers/flash/flash_stm32l4x.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /drivers/flash/flash_stm32l4x.c
 Category: Uninitialized variables
 Function: erase_page
 Component: Drivers
 CID: 203515
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203515:  Uninitialized variables  (UNINIT)
/drivers/flash/flash_stm32l4x.c: 149 in erase_page()
143     
144     	/* Set the PER bit and select the page you wish to erase */
145     	regs->cr |= FLASH_CR_PER;
146     #ifdef FLASH_CR_BKER
147     	regs->cr &= ~FLASH_CR_BKER_Msk;
148     	/* Select bank, only for DUALBANK devices */
>>>     CID 203515:  Uninitialized variables  (UNINIT)
>>>     Using uninitialized value "pages_per_bank".
149     	if (page >= pages_per_bank)
150     		regs->cr |= FLASH_CR_BKER;
151     #endif
152     	regs->cr &= ~FLASH_CR_PNB_Msk;
153     	regs->cr |= ((page % pages_per_bank) << 3);
154     
```

aescolar, 2019-08-18:
@gon1332  this comes from 4fffe790be2 #17098 , could you please take a look at it?

[image]

gon1332, 2019-08-19:
@aescolar Yes I'll look into it.

LeiW000, 2019-08-21:
I reviewed detail of the issue in https://scan9.coverity.com/reports.htm#v32951/p12996. In the latest code, commit 0381e8c can fix this issue. So close this bug.

@aasthagr, thanks for heads-up.

closed 2019-08-21 by LeiW000